sle118/squeezelite-esp32 #7: OTA Request blocked on firefox

thomaspreece, 2019-10-02:
Using Firefox on Windows. The OTA "Check For Updates" button results in a "failed to fetch release history" error. The debugger shows the following error:
```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://api.github.com/repos/sle118/squeezelite-esp32/releases. (Reason: CORS request did not succeed).
```

sle118, 2019-10-02:
Server code needs to add support for Access-Control-Allow-Origin

sle118, 2019-10-03:
@thomaspreece in the meantime, you could try copying/pasting the squeezelite binary from the github release page into the flash url.

thomaspreece, 2019-10-03:
So I got around it by using Chrome. As a side note, pasting into the box also didn't seem to work as I hit a character limit

sle118, 2019-10-05:
Please try the new nvs branch and deepen the issue of the problem is not resolved. Note that the length issue should also be addressed in this branch. Soon if not already; I'm refactoring the backend, and @daduke the frontend to use full JSON instead of htp headers.

closed 2019-10-05 by sle118