sle118/squeezelite-esp32 #14: Always builds 16bit I2S

thomaspreece, 2019-10-26:
I'm not at all experienced with CMake but just going some quick testing, setting CONFIG_I2S_BITS_PER_CHANNEL_24=y in the menuconfig still results in CONFIG_I2S_BITS_PER_CHANNEL=16 due to I believe the following line:
https://github.com/sle118/squeezelite-esp32/blob/7873829c0a23cba36e05caf0002f1dd2a312b52d/main/Kconfig.projbuild#L103

sle118, 2019-10-28:
We currently only support 16 bits, as higher bit rate may not run as reliably on the esp32. This may be worth investigating in case we want to merge back into the core of squeezelite at some point.

@philippe44 could you comment?

philippe44, 2019-10-28:
When I ported squeezelite to make it embedded system friendly, I added internal 16 bits representation for all codecs (it was hard coded 32 bist before). This is the default and validated format used now. 

My squeezelite port is theoritically capable of doing 16 or 32 bits internal, but I've not fully tested it (only quickly on Linux) and using 32 bits for esp32 would limit a lot the buffering length, so I've decided to stick with the 16 bits version

thomaspreece, 2019-10-29:
Ah ok. Might be worth removing that option from the build config then.

closed 2020-04-17 by sle118