sle118/squeezelite-esp32 #18: Compiling and flashing fails

DarkShortie, 2020-01-22:
I wanted to give this very interesting project a try, but right now I'm failing to compile it.
I've used Docker to install and start ESP-IDF.
For the config I've used squeezelite-esp32-I2S-4MFlash-sdkconfig.defaults for testing purposes (old Hifiberry Miniamp which doesn't require i2c is conntected to the esp32).

After entering the Docker container I've issued the commands described in the topic "Building Squeezelite-esp32"
I didn't got the point "for all libraries, add -mlongcalls.", what I'm supposed to do here exactly?

The make flash command fails with 
```
In file included from /workspace/squeezelite-esp32/main/esp_app_main.c:45:0:
/workspace/squeezelite-esp32/components/wifi-manager/wifi_manager.h:46:2: error: #error "RECOVERY_APPLICATION not defined. Defaulting to squeezelite"
 #error "RECOVERY_APPLICATION not defined. Defaulting to squeezelite"
  ^
/workspace/squeezelite-esp32/components/wifi-manager/wifi_manager.h:51:9: note: #pragma message: compiling for squeezelite.
 #pragma message "compiling for squeezelite."
         ^
cc1: warning: unrecognized command line option '-Wno-frame-address'
/workspace/esp-idf/make/component_wrapper.mk:291: recipe for target 'esp_app_main.o' failed
make[1]: *** [esp_app_main.o] Error 1
/workspace/esp-idf/make/project.mk:604: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
```

If I repeat those steps then it fails with
```
App "recovery" version: v0.4.232-v3.1.5-SqueezeAmp4MBFl
CC build/mqtt/esp-mqtt/mqtt_client.o
/workspace/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c: In function 'esp_mqtt_client_init':
/workspace/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:487:9: error: implicit declaration of function 'esp_transport_ssl_set_client_key_password' [-Werror=implicit-function-declaration]
         esp_transport_ssl_set_client_key_password(ssl,
         ^
/workspace/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:504:9: error: implicit declaration of function 'esp_transport_ssl_set_alpn_protocol' [-Werror=implicit-function-declaration]
         esp_transport_ssl_set_alpn_protocol(ssl, (const char **)client->config->alpn_protos);
         ^
/workspace/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
cc1: some warnings being treated as errors
/workspace/esp-idf/make/component_wrapper.mk:291: recipe for target 'esp-mqtt/mqtt_client.o' failed
make[1]: *** [esp-mqtt/mqtt_client.o] Error 1
/workspace/esp-idf/make/project.mk:604: recipe for target 'component-mqtt-build' failed
make: *** [component-mqtt-build] Error 2
```

As an alternative I wanted to flash the release https://github.com/sle118/squeezelite-esp32/releases/tag/v0.4.218-v3.1.5-I2S-4MFlash-master
For this I've downloaded and extracted the zip file.
The flashing attemp fails with this error
```
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "squeezelite.bin" must be a number
```

ithinkido, 2020-01-23:
I can not help with the complinig, but flashing is a two step process. Start with only the recovery bin, ie

Python esptool.py --port COM** --baud 921600  write_flash --flash_mode dio --flash_freq 80m --flash_size detect  0x1000 bootloader/bootloader.bin 0x8000 partitions.bin 0xD000 ota_data_initial.bin 0x10000 recovery.bin

you will find an AP - squeezelite------ 
connect to it ( PW squeezlite ) and enter your wifi data.
after that you can do a firmware update from the web page

philippe44, 2020-01-23:
You have to explicitely do the steps describes in the README, so type
do a 'make defconfig' then
```
find . \( -name "*.cpp" -o -name "*.c" -o -name "*.h" \) -type f -print0 | xargs -0 grep -l "RECOVERY_APPLICATION" | xargs touch
export PROJECT_NAME="recovery" 
make -j4 all EXTRA_CPPFLAGS='-DRECOVERY_APPLICATION=1'
```
Then do the 2nd step as describe following what I've copied/pasted above

DarkShortie, 2020-01-24:
Thanks for your hints.
I managed to flash the precompiled releases by changing the command and then doing an OTA like suggested.

To get it compiling I've executed `git submodule update --init --recursive` in /workspace/esp-idf

closed 2020-01-24 by DarkShortie