sle118/squeezelite-esp32 #20: Support for Volume UP/DOWN Buttons on iOS / Android

cribskip, 2020-04-26:
Using the buttons on my (Android) Phone I've noticed that the volume buttons are not showing any effect. Paired with my Windows Laptop or using LMS, volume control works just fine.

So I've looked in the code and found this comment:
https://github.com/sle118/squeezelite-esp32/blob/7a44e379edc3fb04434b52af49a5f26a650911fc/components/driver_bt/bt_app_sink.c#L91

And this upstream issue with the volume buttons:
https://github.com/espressif/esp-idf/issues/4438#issuecomment-566483768

The serial difference between Android and Windows are these. Android button presses are not even shown:

    Android:
    E (197000) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
    I (22330) BT_AV: A2DP connection state: Connecting, [a8:9c:ed:90:fa:e5]
    I (22560) BT_AV: A2DP audio stream configuration, codec type 0
    [00:03:37.154601] bt_sink_cmd_handler:155 Setting BT sample rate 44100
    I (22560) BT_AV: Configure audio player 21-15-2-35
    I (22570) BT_AV: Audio player configured, sample rate=44100
    W (22780) BT_APPL: new conn_srvc id:19, app_id:0
    I (22780) BT_AV: A2DP connection state: Connected, [a8:9c:ed:90:fa:e5]
    I (23110) RCCT: AVRC conn_state evt: state 1, [a8:9c:ed:90:fa:e5]
    I (23140) RCCT: remote rn_cap: count 6, bitmask 0xe26
    I (23170) RCCT: AVRC metadata rsp: attribute id 0x1,
    I (23180) RCCT: AVRC metadata rsp: attribute id 0x2,
    I (23180) RCCT: AVRC metadata rsp: attribute id 0x4,
    I (23190) RCCT: AVRC metadata rsp: attribute id 0x40,
    I (23410) RCTG: AVRC register event notification: 13, param: 0x0

    I (64950) BT_AV: Playback status changed: 0x2
    W (64960) BT_AV: Not yet in BT connected mode: 0x2
    I (65080) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

    W (65080) BT_APPL: new conn_srvc id:19, app_id:1
    I (65090) BT_AV: A2DP audio state: Started
    [00:04:19.696679] bt_sink_cmd_handler:130 BT sink started
    [00:04:19.697994] bt_sink_cmd_handler:141 BT playing
    I (65100) BT_AV: Track changed
    I (65110) BT_AV: Playback status changed: 0x1
    [00:04:19.732882] output_thread_i2s:406 Output state is 2
    I (65160) RCCT: AVRC metadata rsp: attribute id 0x1, She Knows (feat. Amber Coffman & Cults) |
    I (65170) RCCT: AVRC metadata rsp: attribute id 0x2, J. Cole
    I (65180) RCCT: AVRC metadata rsp: attribute id 0x4, Born Sinner (Deluxe Version)
    I (65180) RCCT: AVRC metadata rsp: attribute id 0x40, 296545

    I (83770) BT_AV: Track changed
    I (83780) BT_AV: Playback status changed: 0x2
    [00:04:38.375519] bt_sink_cmd_handler:151 BT paused, just silence
    I (83860) RCCT: AVRC metadata rsp: attribute id 0x1, Mona Lisa
    I (83860) RCCT: AVRC metadata rsp: attribute id 0x2, Kool savas
    I (83870) RCCT: AVRC metadata rsp: attribute id 0x4,
    I (83880) RCCT: AVRC metadata rsp: attribute id 0x40, 200568
    I (84370) BT_AV: Playback status changed: 0x0
    [00:04:38.964991] bt_sink_cmd_handler:147 BT stopped
    [00:04:38.982234] output_thread_i2s:406 Output state is 0
    I (84630) BT_AV: Playback status changed: 0x2
    [00:04:39.232802] bt_sink_cmd_handler:151 BT paused, just silence
    I (84650) BT_AV: Track changed
    I (84730) BT_AV: Playback status changed: 0x1
    [00:04:39.331063] bt_sink_cmd_handler:141 BT playing
    I (84740) RCCT: AVRC metadata rsp: attribute id 0x1, Mona Lisa
    I (84740) RCCT: AVRC metadata rsp: attribute id 0x2, Kool savas
    I (84750) RCCT: AVRC metadata rsp: attribute id 0x4,
    [00:04:39.353760] output_thread_i2s:406 Output state is 2
    I (84760) RCCT: AVRC metadata rsp: attribute id 0x40, 200573

    I (101040) BT_AV: Playback status changed: 0x2
    [00:04:55.641095] bt_sink_cmd_handler:151 BT paused, just silence
    E (102680) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
    I (102680) RCCT: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
    I (102680) BT_AV: A2DP audio state: Stopped
    [00:04:57.282075] bt_sink_cmd_handler:136 BT sink stopped
    I (102700) BT_AV: A2DP connection state: Disconnected, [a8:9c:ed:90:fa:e5]
    [00:04:57.325974] output_thread_i2s:406 Output state 

    Windows:
    E (197000) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
    I (197000) BT_AV: A2DP connection state: Connecting, [04:ed:33:8b:73:23]
    I (197310) BT_AV: A2DP audio stream configuration, codec type 0
    [00:06:31.907957] bt_sink_cmd_handler:155 Setting BT sample rate 44100
    I (197310) BT_AV: Configure audio player 21-15-2-35
    I (197320) BT_AV: Audio player configured, sample rate=44100
    I (197450) RCTG: AVRC conn_state evt: state 1, [04:ed:33:8b:73:23]
    W (197690) BT_APPL: new conn_srvc id:19, app_id:0
    I (197690) BT_AV: A2DP connection state: Connected, [04:ed:33:8b:73:23]
    I (197780) RCTG: AVRC register event notification: 13, param: 0x0
    I (197900) RCCT: AVRC conn_state evt: state 1, [04:ed:33:8b:73:23]
    I (197900) RCCT: AVRC remote features 24b, TG features 11
    I (197910) RCTG: AVRC remote features 24b, CT features 3
    I (197990) RCCT: remote rn_cap: count 6, bitmask 0xf06
    I (198170) RCCT: AVRC metadata rsp: attribute id 0x1, Bauer sucht Frau - Schäfer Heinricht sucht seine Traumfrau | Switch Reloaded Classics #reupload
    I (198190) RCCT: AVRC metadata rsp: attribute id 0x2, Switch reloaded
    E (200930) BT_APPL: bta_dm_pm_btm_status hci_status=36
    I (247560) BT_AV: Playback status changed: 0x1
    W (247560) BT_AV: Not yet in BT connected mode: 0x1
    I (248580) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

    W (248580) BT_APPL: new conn_srvc id:19, app_id:1
    I (248590) BT_AV: A2DP audio state: Started
    [00:07:23.194328] bt_sink_cmd_handler:130 BT sink started
    [00:07:23.195999] bt_sink_cmd_handler:141 BT playing
    [00:07:23.240197] output_thread_i2s:406 Output state is 2
    I (253910) RCTG: AVRC set absolute volume: 2%
    I (253910) RCTG: Volume is set by remote controller 2%

    [00:07:28.509953] volume:155 TAS57xx volume (L:0 R:0)
    I (254830) RCTG: AVRC set absolute volume: 3%
    I (254830) RCTG: Volume is set by remote controller 3%

    [00:07:29.424431] volume:155 TAS57xx volume (L:3 R:3)
    I (255590) RCTG: AVRC set absolute volume: 6%
    I (255590) RCTG: Volume is set by remote controller 6%

    [00:07:30.188233] volume:155 TAS57xx volume (L:16 R:16)
    [00:07:30.333991] discover_server:805 sending discovery
    I (257020) RCTG: AVRC set absolute volume: 3%
    I (257020) RCTG: Volume is set by remote controller 3%

    [00:07:31.619630] volume:155 TAS57xx volume (L:3 R:3)
    I (259960) BT_AV: Playback status changed: 0x2

Right now, I'm at v0.5.543-IDFv3.1.5-SqueezeAmp8M

philippe44, 2020-04-29:
I've not made work about that in a while, but I remember that conclusion that Android has that strange no up/down support. Does not mean you can't change volume, but it's an "absolute" level set, not an up/down. Try to change the volume through the UI control

sle118, 2020-04-30:
Which branch does this apply to?

philippe44, 2020-04-30:
all of them - vol up/down is not supported by Android

cribskip, 2020-05-28:
There is no way to change the volume from android for me. Neither with buttons nor with the ui 'volume bar'.

Only solution to use squeezeamp as a 'Bluetooth speaker' was to add a rotary encoder.

sle118, 2020-08-28:
@cribskip could you please try the bleeding edge release? We are now in esp-idf v4.0, which should support absolute volume and we already have code in to support that.

sle118, 2020-08-29:
Flagged as enhancement. None of my tablets are misbehaving, and none of my friend's either so I'm not sure how I can test for myself. @cribskip what kind of tablet are you using? Did you try other devices?

sle118, 2020-09-06:
Closing due to inactivity

closed 2020-09-06 by sle118