sle118/squeezelite-esp32 #21: Compile issue under windows

Phil1pp, 2020-04-27:
I am trying to compile the cmake branch on windows 10 with idf 4.0.
idf.py makemenuconfig is working fine but after idf.py build I get the following error:

```php
C:\esp-idf-40\projects\squeezelite-esp32>idf.py build
Checking Python dependencies...
Python requirements from C:\esp-idf-40\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\esp-idf-40\projects\squeezelite-esp32\build
Executing "ninja all"...
[2/10] cmd.exe /C "cd /D C:\esp-idf-40\projects\squeezeli.../projects/squeezelite-esp32/generate_debug_scripts.cmake"
Generating debug script files
[3/10] Performing build step for 'bootloader'
ninja: no work to do.
[7/8] Generating binary image from built executable
esptool.py v2.8
Generated C:/esp-idf-40/projects/squeezelite-esp32/build/recovery.bin
[8/8] Linking CXX executable squeezelite.elf
FAILED: squeezelite.elf
cmd.exe /C "cmd.exe /C "cd /D C:\esp-idf-40\projects\squeezelite-esp32\build && xtensa-esp32-elf-objcopy --weaken-symbol esp_app_desc C:/esp-idf-40/projects/squeezelite-esp32/build/esp-idf/app_update/libapp_update.a && xtensa-esp32-elf-objcopy --globalize-symbol find_command_by_name C:/esp-idf-40/projects/squeezelite-esp32/build/esp-idf/console/libconsole.a && cd C:\esp-idf-40\projects\squeezelite-esp32\build" && C:\esp-idf-tools\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\squeezelite.elf.rsp  -o squeezelite.elf  && cmd.exe /C "cd /D C:\esp-idf-40\projects\squeezelite-esp32\build && C:\esp-idf-tools\tools\cmake\3.13.4\bin\cmake.exe -E echo "Generating C:/esp-idf-40/projects/squeezelite-esp32/build/squeezelite.bin" && C:\esp-idf-tools\python_env\idf4.0_py3.8_env\Scripts\python.exe C:/esp-idf-40/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB --elf-sha256-offset 0xb0 -o C:/esp-idf-40/projects/squeezelite-esp32/build/squeezelite.bin squeezelite.elf""
c:/esp-idf-tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../components/codecs/lib/libalac.a(alac_wrapper.o):(.literal+0xc): undefined reference to `_Unwind_Resume'
c:/esp-idf-tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../components/codecs/lib/libalac.a(alac_wrapper.o): in function `alac_create_decoder':
D:\Developments\msys32\home\esp32\squeezelite\alac/alac_wrapper.cpp:38: undefined reference to `_Unwind_Resume'
c:/esp-idf-tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../components/codecs/lib/libalac.a(alac_wrapper.o):(.eh_frame+0x12): undefined reference to `__gxx_personality_v0'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

It seems like the audio libraries are compiled with a different compiler version.
How can I fix that?
Whats the easiest way to get a default built with SPDIF enabled?

philippe44, 2020-04-29:
I've compiled the codecs with 5.2.0. 
I've kept an older IDF version and so far only @sle118 has moved to 4.0 including moving to gcc 8.x, but I don't think have had problem using codecs compiled with 5.2.0, so your issue probably lies somewhere else, with the combination of 5.2.0 and IDC 4.0

sle118, 2020-04-30:
Please report which version of the esp-idf and espressif tools you are using. The make files are being used in both Windows 10 (daily development environment under eclipse) as well as Linux (Jenkins build system). I've never seen this issue happening before.

First try to compile with one of the build scripts before going on menuconfig. If you are still not successful, I'll see what I can do.

sle118, 2020-04-30:
Also have a look at the issue below, which recommends updating the esp-idf modules recursively.

https://github.com/espressif/esp-idf/issues/4503

sle118, 2020-05-06:
Closing due to inactivity

closed 2020-05-06 by sle118