sle118/squeezelite-esp32 #22: Compile problem

ppociechaCreateIt, 2020-04-29:
Hi, I have problem with compile this on Ubuntu ()Windows subsystem). I have a lot of errors. 
Firstly I make everything by this instruction (https://www.instructables.com/id/ESP32-Development-on-Windows-Subsystem-for-Linux/). In esp-idf I change toolchain to 5.2.0 make defconfig, and when 
I run make I have a lot of error firstly is:

In file included from /home/pawlak/esp/esp-idf/components/heap/heap_private.h:20:0,
                 from /home/pawlak/esp/esp-idf/components/heap/heap_caps_init.c:14:
/home/pawlak/esp/esp-idf/components/heap/heap_caps_init.c: In function 'heap_caps_add_region_with_caps':
/home/pawlak/esp/esp-idf/components/heap/multi_heap_platform.h:32:20: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
         if((PLOCK) != NULL) {                               \
                    ^
/home/pawlak/esp/esp-idf/components/heap/heap_caps_init.c:230:5: note: in expansion of macro 'MULTI_HEAP_LOCK'
     MULTI_HEAP_LOCK(&registered_heaps_write_lock);
     ^
/home/pawlak/esp/esp-idf/components/heap/multi_heap_platform.h:39:21: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
         if ((PLOCK) != NULL) {                              \
                     ^
/home/pawlak/esp/esp-idf/components/heap/heap_caps_init.c:232:5: note: in expansion of macro 'MULTI_HEAP_UNLOCK'
     MULTI_HEAP_UNLOCK(&registered_heaps_write_lock);
     ^

nextone is:
/mnt/d/squeezelite-esp32-master/main/console.c:122:36: error: 'CONFIG_CONSOLE_UART_NUM' undeclared (first use in this function)

philippe44, 2020-04-29:
You have to use a certain version of the IDF. The jenkins build that @sle118 is using a version that is a tiny bit different than the one I'm using locally https://github.com/espressif/esp-idf/tree/16014079f9486b34679ee6c6eb6f554df4e3c313 but I can't find right now which one jenkins uses. Still, there should only be one error when you compile telnet. Grab our config file as well

ppociechaCreateIt, 2020-06-09:
@philippe44 thx a lot - that was the problem.

closed 2020-06-21 by daduke