sle118/squeezelite-esp32 #23: Make prefs conditional to prevent error being logged

michaelherger, 2020-05-05:
On https://github.com/sle118/squeezelite-esp32/blob/4f324b90d7d0534b8bf0969467ab32cc0f1d5e57/plugin/SqueezeESP32/PlayerSettings.pm#L33 you define what prefs to handle in the player settings. But many of these are not define in the template unless you have a display connected. Therefore you should make these conditional:

```diff
diff --git a/plugin/SqueezeESP32/PlayerSettings.pm b/plugin/SqueezeESP32/PlayerSettings.pm
index 9b331d8..33c275a 100644
--- a/plugin/SqueezeESP32/PlayerSettings.pm
+++ b/plugin/SqueezeESP32/PlayerSettings.pm
@@ -30,7 +30,8 @@ sub page {
 
 sub prefs {
        my ($class, $client) = @_;
-       my @prefs = qw(width small_VU spectrum artwork eq);
+       my @prefs;
+       shift @prefs, qw(width small_VU spectrum artwork eq) if $client->displayWidth;
        return ($prefs->client($client), @prefs);
 }
```

philippe44, 2020-05-05:
but eq is always there, so I assume I should have
```
   my @prefs = qw (eq);
   shift @prefs, qw(width small_VU spectrum artwork) if $client->displayWidth;
```

michaelherger, 2020-05-05:
You’re right. I’ll have a PR with a few more changes later today.

philippe44, 2020-05-05:
That's scary :-)

sle118, 2020-05-05:
indeed, scary!

commit be00683b (2020-05-07): Merge pull request #24 from michaelherger/fix-no-display-use-case-and-some

Fix #23 - fix some issues when SqueezeAmp doesn't have a display
closed 2020-05-07 by philippe44