sle118/squeezelite-esp32 #30: Add Ogg/Flac support

ralph-irving, 2020-05-17:
Attached is a patch that adds ogg/flac support.

Unfortunately, I don't have an esp32 dev environment so I can't build and test it on my squeezeamp.  I added ogg/flac to squeezelite in August of last year and so far it's been solid.

[oggflac.patch.txt](https://github.com/sle118/squeezelite-esp32/files/4640684/oggflac.patch.txt)

I didn't create a PR because I couldn't build and test it..  Hopefully it works for squeezeamp....I've been enjoying mine!

philippe44, 2020-05-20:
Thanks, I did not see the patch till today. I've tried it but unfortunately it's not working, the streams end in FLAC__STREAM_DECODER_UNINITIALIZED although the ogg init function is called. I'll have to investigate better.

One thing though, at the risk of looking like a purist :-)

Why checking for the 'o' exception in decode.c which normally does not know anything about the underlying codec it's calling. Knowing that codec->open is always called by decode.c, whether we switch codec or have flac to flac transition, why not tracking inside flac.c the type of codec and if it's a flac to flac but with a container change, close the codec and re-open a new one?

ralph-irving, 2020-05-20:
No worries.  Is your flac library for esp32 compiled with ogg support enabled?  It doesn't have to be to work with only flac.
Thanks for reviewing the code. I'm always happy to have second option.  I don't remember why I did the codec close/open in decode.c.  I'll take another look at it with your suggest in mind.

ralph-irving, 2020-05-20:
I've removed the changes to decode.c and refactored them into flac.c  I'll continue to test and will post an updated patch in the next couple days. Hopefully you'll figure out why you're getting the uninitialized error.

ralph-irving, 2020-05-20:
Here's the refactored patch.  I haven't committed it to github yet.

[oggflac-v2.patch.txt](https://github.com/sle118/squeezelite-esp32/files/4658289/oggflac-v2.patch.txt)

philippe44, 2020-05-21:
@ralph-irving - thanks, I've added something a tiny bit different because I needed the container flag to be part of the decoder struct so that it would work on my bridges where I need multiple instances of the same code

philippe44, 2020-05-22:
One thing BTW that amazed me is the amount of CPU used by OggFlac. Moved from ~20% of one xtensa core for flac-only to 70% for OggFlac. This is bad :-(

ralph-irving, 2020-05-22:
I've been trying to keep our codec sources as close as possible and have applied your additional ogf changes.  I hadn't noticed the excessive cpu usage before.  But I'm seeing ~60% cpu on i386 where flac uses less than 1%.   I know it's not comparing the same thing but playing an ogf stream on the touch, cpu is around 25%.

ralph-irving, 2020-05-22:
I think it's the icy meta data in the radio paradise stream.  There's metadata about every 5 seconds.  Try the other two test urls I included.  CPU only ~5% 

[08:46:03.180402] stream_thread:300 icy meta: len: 48
StreamTitle='Ashley MacIsaac - Sleepy Maggie';
[08:46:03.180470] sendSTAT:189 STAT: STMt
[08:46:03.180544] sendMETA:233 META
[08:46:04.233954] sendSTAT:189 STAT: STMt
[08:46:05.140301] process:521 strm
[08:46:05.140340] process_strm:274 strm command t
[08:46:05.140354] sendSTAT:189 STAT: STMt
[08:46:06.341223] sendSTAT:189 STAT: STMt
[08:46:07.533134] stream_thread:300 icy meta: len: 48
StreamTitle='Ashley MacIsaac - Sleepy Maggie';
[08:46:07.560820] sendSTAT:189 STAT: STMt
[08:46:07.560894] sendMETA:233 META
[08:46:08.592606] sendSTAT:189 STAT: STMt
[08:46:09.343033] process:521 strm
[08:46:09.343064] process_strm:274 strm command t
[08:46:09.343074] sendSTAT:189 STAT: STMt
[08:46:10.394116] sendSTAT:189 STAT: STMt
[08:46:11.444685] sendSTAT:189 STAT: STMt
[08:46:11.886764] stream_thread:300 icy meta: len: 48
StreamTitle='Ashley MacIsaac - Sleepy Maggie';
[08:46:11.927550] sendMETA:233 META
[08:46:12.945810] sendSTAT:189 STAT: STMt
[08:46:13.000863] process:521 strm
[08:46:13.000903] process_strm:274 strm command t
[08:46:13.000941] sendSTAT:189 STAT: STMt
[08:46:14.146382] sendSTAT:189 STAT: STMt
[08:46:15.197002] sendSTAT:189 STAT: STMt
[08:46:16.197526] sendSTAT:189 STAT: STMt
[08:46:16.239998] stream_thread:300 icy meta: len: 48
StreamTitle='Ashley MacIsaac - Sleepy Maggie';

philippe44, 2020-05-22:
yes you're right it's a bit better, like 50% CPU instead of 70%, but normally my worse codec is ~30% and flac is 17%

philippe44, 2020-05-22:
BTW, I saw you are adopting my changes of being able to use 4 bytes per frame in squeezelite, so first I'm flattered :-) but also, do you have a special project in mind? (just asking to know if I could help)

ralph-irving, 2020-05-23:
Your alac, opus and voris updates already had the 4 byte frame changes so it made sense to add the change to the rest so squeezelite could actually be built using 4 byte frames.  When I first started looking at updating the radio firmware I investigated using squeezelite and jivelite but you loose several key features provided by squeezeplay, like backup alarm, ui sound effects and support for the sounds and effects app.

philippe44, 2020-05-23:
The 4 bytes overall was necessary for the esp32 project. 

From a CPU point of view, I was afraid that 32 bits per sample would be too much and regardless, memory was too tight to allow proper buffering at 32 bits. It also allowed to build a reasonably low CPU fixed point resample as soxr was too demanding. With 32 bits, resampling would have been much more complex as the xtensa is a 32 bits processor with limited possibility of 64 bits arithmetic. It has a FPU, but it's not super efficient.

And ... my ears can't spot the difference :-)

ralph-irving, 2020-05-24:
I can't hear the difference between 16 and 24 or 32 bit either.
I was following the esp32 project when you added the 16bit resampler and thought I could use it with the radio firmware as well since it currently uses the "crappy" alsa resampler to anything not 44100/48000.  For now that's all on the back burner as my current radio firmware is working very well.

ralph-irving, 2020-05-24:
Thanks for adding ogg/flac support.

closed 2020-05-24 by ralph-irving