sle118/squeezelite-esp32 #33: NVS key/value documentation

dionisor, 2020-05-18:
Is there any documentation for key/value pair?

philippe44, 2020-05-21:
no, except in the code :-) + the README that explains some of them (the most user-important ones)

sle118, 2020-05-29:
I am working on a better front end for platform configuration. There is a new system tab available in the master-CMake branch, where a subset of the terminal command lines are exposed. I am going to expand the offering with configuration actions for Bluetooth source, etc, making the user relevant NVS documented and validated with a front end. NVS editor has always been the last resort option to fixing configuration and there is no plan to document individual NVS parameters.

https://github.com/sle118/squeezelite-esp32/commit/293d08deec74113df02f2c9c27e0d67944085861

closed 2020-05-29 by sle118