sle118/squeezelite-esp32 #34: build problem

ElFishi, 2020-05-23:
Still trying to build my own ...
Ubuntu 20.04 on Windows Subsystem for Linux
installed idf by
`git clone -b v4.0 --recursive https://github.com/espressif/esp-idf.git`
and squeezelite by 
`git clone -b bleeding.edge.59-v4.0-I2S-4MFlash-master-cmake --recursive https://github.com/sle118/squeezelite-esp32.git`

`idf.py build` yields

```
Python requirements from /home/user/esp/esp-idf/requirements.txt are satisfied.
[1/10] cd /home/user/esp/squeezelite-esp32/build && /usr/bin/cmake -P /home/user/esp/squeezelite-esp32/generate_debug_scripts.cmake
Generating debug script files
[2/10] Generating squeezelite_src.c, BUILD
[3/10] Performing build step for 'bootloader'
ninja: no work to do.
[4/8] Generating ld/esp32.project.ld
[5/8] Linking CXX executable recovery.elf
[6/8] Building C object CMakeFiles/squeezelite.elf.dir/squeezelite_src.c.obj
[7/8] Generating binary image from built executable
esptool.py v2.8
Generated /home/user/esp/squeezelite-esp32/build/recovery.bin
[8/8] Linking CXX executable squeezelite.elf
FAILED: squeezelite.elf
cd /home/user/esp/squeezelite-esp32/build && xtensa-esp32-elf-objcopy --weaken-symbol esp_app_desc /home/user/esp/squeezelite-esp32/build/esp-idf/app_update/libapp_update.a && xtensa-esp32-elf-objcopy --globalize-symbol find_command_by_name /home/user/esp/squeezelite-esp32/build/esp-idf/console/libconsole.a && cd /home/user/esp/squeezelite-esp32/build && /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/squeezelite.elf.dir/squeezelite_src.c.obj  -o squeezelite.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/freertos/libfreertos.a  esp-idf/vfs/libvfs.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/unity/libunity.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/tools/libtools.a  esp-idf/platform_config/libplatform_config.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/telnet/libtelnet.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/main/libmain.a  esp-idf/audio/libaudio.a  esp-idf/codecs/libcodecs.a  esp-idf/driver_bt/libdriver_bt.a  esp-idf/esp-dsp/libesp-dsp.a  esp-idf/raop/libraop.a  esp-idf/squeezelite/libsqueezelite.a  esp-idf/app_squeezelite/libapp_squeezelite.a  -Wl,--cref -Wl,--Map=/home/user/esp/squeezelite-esp32/build/squeezelite.map  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/telnet/libtelnet.a  -Wl,--undefined=GDS_DrawPixelFast  esp-idf/app_squeezelite/libapp_squeezelite.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/nghttp/libnghttp.a  esp-idf/squeezelite/libsqueezelite.a  esp-idf/audio/libaudio.a  ../components/audio/lib/libesp_processing.a  -u pow  -u cos  -u sin  -u sqrt  esp-idf/driver_bt/libdriver_bt.a  esp-idf/bt/libbt.a  -L/home/user/esp/esp-idf/components/bt/controller/lib  -lbtdm_app  esp-idf/esp-dsp/libesp-dsp.a  esp-idf/raop/libraop.a  esp-idf/mdns/libmdns.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/platform_config/libplatform_config.a  esp-idf/console/libconsole.a  esp-idf/json/libjson.a  esp-idf/tools/libtools.a  -u _ZdlPv  -u _Znwj  esp-idf/codecs/libcodecs.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/freertos/libfreertos.a  esp-idf/vfs/libvfs.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/freertos/libfreertos.a  esp-idf/vfs/libvfs.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a  /home/user/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a  /home/user/esp/esp-idf/components [... cut ...]
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/codecs/lib/libFLAC.a(lpc.o): in function `FLAC__lpc_compute_best_order':
lpc.c:(.text+0xa0d9): undefined reference to `log'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Checking Python dependencies...
Executing action: all (aliases: build)
Running ninja in directory /home/user/esp/squeezelite-esp32/build
Executing "ninja all"...
ninja failed with exit code 1
```
help?

sle118, 2020-05-29:
Please pull the latest changes from master CMake. I did make some corrections to address similar issues

ElFishi, 2020-05-30:
tried `git clone -b bleeding.edge.62-v4.0-SqueezeAmp8MBFlash-master-cmake --recursive https://github.com/sle118/squeezelite-esp32.git`, but still get a similar result
```
esptool.py v2.8
Generated /home/user/esp/squeezelite-esp32/build/recovery.bin
[1264/1264] Linking CXX executable squeezelite.elf
FAILED: squeezelite.elf
cd /home/user/esp/squeezelite-esp32/build && xtensa-esp32-elf-objcopy --weaken-symbol esp_app_desc /home/user/esp/squeezelite-esp32/build/esp-idf/app_update/libapp_update.a && xtensa-esp32-elf-objcopy --globalize-symbol find_command_by_name /home/user/esp/squeezelite-esp32/build/esp-idf/console/libconsole.a && cd /home/user/esp/squeezelite-esp32/build && /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/squeezelite.elf.dir/squeezelite_src.c.obj  -o squeezelite.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/freertos/libfreertos.a  esp-idf/vfs/libvfs.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/unity/libunity.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/tools/libtools.a  esp-idf/platform_config/libplatform_config.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/telnet/libtelnet.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/main/libmain.a  esp-idf/audio/libaudio.a  esp-idf/codecs/libcodecs.a  esp-idf/driver_bt/libdriver_bt.a  esp-idf/esp-dsp/libesp-dsp.a  esp-idf/raop/libraop.a  esp-idf/squeezelite/libsqueezelite.a  esp-idf/app_squeezelite/libapp_squeezelite.a  -Wl,--cref -Wl,--Map=/home/user/esp/squeezelite-esp32/build/squeezelite.map  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/platform_console/libplatform_console.a  esp-idf/squeezelite-ota/libsqueezelite-ota.a  esp-idf/wifi-manager/libwifi-manager.a  esp-idf/telnet/libtelnet.a  -Wl,--undefined=GDS_DrawPixelFast  esp-idf/app_squeezelite/libapp_squeezelite.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/nghttp/libnghttp.a  esp-idf/squeezelite/libsqueezelite.a  esp-idf/audio/libaudio.a  ../components/audio/lib/libesp_processing.a  -u pow  -u cos  -u sin  -u sqrt  esp-idf/driver_bt/libdriver_bt.a  esp-idf/bt/libbt.a  -L/home/user/esp/esp-idf/components/bt/controller/lib  -lbtdm_app  esp-idf/esp-dsp/libesp-dsp.a  esp-idf/raop/libraop.a  esp-idf/mdns/libmdns.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/services/libservices.a  esp-idf/display/libdisplay.a  esp-idf/platform_config/libplatform_config.a  esp-idf/console/libconsole.a  esp-idf/json/libjson.a  esp-idf/tools/libtools.a  -u _ZdlPv  -u _Znwj  esp-idf/codecs/libcodecs.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp- [... cut ...]
Running cmake in directory /home/user/esp/squeezelite-esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /home/user/esp/squeezelite-esp32"...
Running ninja in directory /home/user/esp/squeezelite-esp32/build
Executing "ninja all"...
ninja failed with exit code 1
```
where is my error?
Attached please find the full output of the `idf.py` build command 
[idf-build.log](https://github.com/sle118/squeezelite-esp32/files/4706517/idf-build.log)