sle118/squeezelite-esp32 #35: Clicks/Pop in alac decoder

ralph-irving, 2020-05-24:
@philippe44  I've been trying to track down an issue with tiny audio dropouts/breaks when playing alac files with the apple alac decoder without success.  I'm experiencing the problem with the current squeezeamp bleeding.edge.59.master build as well as with previous idf v3.1.5 builds.
I can reproduce the same issue with squeezlite linux i386 and armv6hf builds tool, so it's not squeezeamp specific.  Would you be able to help? I can provide alac files to test if needed.  Thanks.

philippe44, 2020-05-24:
If you have specific files, that would really help. 

There was a "home made" alac decoder that I started to use at first in squeezeesp/amp for the AirPlay (shairport-like) client and I had a *lot* of these clicks. I went to a great deal of tests to identify the issue up to sending fake random+hash alac packets so that I could check for transport layer issues at the receiving edge, but I never got any transport problem. Then in desperation I moved to the apple alac decoder and the problem 99% went away, but it's true that I still hear from time to time these clicks.

So I really don't know where the issue comes from. Now is this long dropout/breaks or little "clicks" that you are hearing?

ralph-irving, 2020-05-25:
It's only little clicks and I find it's a lot easier to notice with headphones but they are still obvious with speakers as well.  

Interesting that some alac decoder implementations were better than others.  I'm using https://github.com/TimothyGu/alac but  I think I still might be using your alac repo sources as well.  Will check and confirm.

I've sent you an email with 4 alac files that exhibit the clicks.

ralph-irving, 2020-05-25:
Confirmed I am still using your alac repo for my armv6hf build all others are using the one from my last post.
Which alac decoder sources are you using?  If I'm going to rebuild the alac library for armv6hf i should use the same one as you.

philippe44, 2020-05-25:
I'm using the same now, but in the past I was using this one https://github.com/mikebrady/shairport-sync/blob/master/alac.c and this is where I had much clicks, but I only noticed them on my SqueezeESP. I'm still using it in ShairTunes and my AirConnect applications but I never had clicks. On AirPlay Bridge I've move to Apple's ALAC (with the above mentionned wrapper)

ralph-irving, 2020-05-25:
Now I remember why I wasn't using https://github.com/TimothyGu/alac for armv6hf.  That library gets a crazy value for the sample rate 1152122880  and segfaults a bit later even though the output device is opened correctly at 44100.

[log.txt](https://github.com/sle118/squeezelite-esp32/files/4678718/log.txt)

Doesn't matter for this discussion.  I'm using the same alac lib as you on my i386 linux systems which exhibits the clicks same as squeezeamp.

sle118, 2020-08-27:
Is this still an issue, and if so is it fixable?

ralph-irving, 2020-08-27:
v0.5.635 still exhibits the issue.  I will update to current and report back.

sle118, 2020-08-29:
@philippe44 is this something that can eventually be fixed?

philippe44, 2020-08-30:
I've tried a lot but I cannot find a reproductible enough use case. It's a really weird one and when I looked to a solution, I build a lot of test code that eliminated transmission errors (i.e. I've sent a ton of pseudo-random frames and they always matched, over millions) so it's probably something funny with the decoder but I've not been able to find out what so far

ralph-irving, 2021-01-15:
Thanks for tracking down this issue.  I haven't heard any pops/clicks in alac files with dev.552.

closed 2021-01-15 by ralph-irving