sle118/squeezelite-esp32 #38: amp GPIO flickers

ElFishi, 2020-06-03:
Both in v0.5.565-IDFv3.1.5-I2S-4MFlash and in bleeding.edge.62-I2S-4MFlash the amp GPIO switches on during boot and off again if the player is off in LMS. This is a bit of a nuisance as it can cause a turn-on thump.
Can squeezelite-esp32 be made in a way that leaves the GPIO low during boot if they player is off?
Thanks

philippe44, 2020-08-28:
Are you sure? It boots OFF and if the timer for auto-off is set, it stays OFF. If the timer is not set, then it starts OFF, goes ON then upon registration with an LMS server, the server sets the value according to the memorized status of the player. This is the usual behavior of squeezelite which is to start ON until the server tells you otherwise. Maybe you have not set a auto-off value (-C option) and that explains.

sle118, 2020-08-28:
Could you elaborate on your setup? More specifically, which gpio did you use?

ElFishi, 2020-08-28:
26=amp
I had indeed not set the -C option.
With -C set the amp stays off during boot.
Problem solved, thx.

closed 2020-08-28 by ElFishi