sle118/squeezelite-esp32 #39: bleeding.edge.62 crashes

ElFishi, 2020-06-03:
When I switch between sources (radio to fileserver or the other way around) squeezelite sometimes crashes. I get the following log
```
[00:03:30.260530] grfs_handler:712 scroller current size 1412 (w:225)
[00:03:30.560642] codec_open:279 codec open: 'm'
assertion "sockets[i].select_waiting == 0" failed: file "/var/lib/jenkins/workspace/v4.0/components/lwip/lwip/src/api/sockets.c", line 565, function: alloc_socket
abort() was called at PC 0x40218df3 on core 1

ELF file SHA256: 5a8e17b9e5a33c24ea458f7184ce8b2abe4ba26813ff62f8c55f2fdc11b543b5

Backtrace: 0x40095868:0x3ffcea00 0x40095bd1:0x3ffcea20 0x40218df3:0x3ffcea40 0x401a535d:0x3ffcea70 0x401a6cbd:0x3ffceaa0 0x400ea40f:0x3ffceac0 0x400e8c19:0x3ffceb00 0x400e9527:0x3ffceb30 0x400e507e:0x3ffcec10 0x400d947f:0x3ffcef40 0x400d41a1:0x3ffcef60

[0;32mI (13933) esp_core_dump_uart: Press Enter to print core dump to UART...[0m
[0;32mI (13940) esp_core_dump_uart: Print core dump to uart...[0m
[0;32mI (13946) esp_core_dump_common: Found tasks: (27)![0m
================= CORE DUMP START =================
tFQAAAEAAAAWAAAAcAEAAA==
```

ElFishi, 2020-06-05:
switching from radio to file I experienced another crash with identical log

ElFishi, 2020-06-23:
bleeding.edge.68
```
[00:01:58.838487] codec_open:279 codec open: 'm'
[00:01:58.839617] codec_open:296 closing codec: 'a'
assertion "sockets[i].select_waiting == 0" failed: file "/var/lib/jenkins/workspace/v4.0/components/lwip/lwip/src/api/sockets.c", line 565, function: alloc_socket
abort() was called at PC 0x40219ac3 on core 1

ELF file SHA256: 00d21d2d1ed6d946e2d69abae260da76af99d4cabcf4ce3176d9d17e70b309f9

Backtrace: 0x40095868:0x3ffce820 0x40095bd1:0x3ffce840 0x40219ac3:0x3ffce860 0x401a585d:0x3ffce890 0x401a71bd:0x3ffce8c0 0x400ea413:0x3ffce8e0 0x400e8c1d:0x3ffce920 0x400e952b:0x3ffce950 0x400e5082:0x3ffcea30 0x400d9483:0x3ffced60 0x400d41a5:0x3ffced80

(14221) esp_core_dump_uart: Press Enter to print core dump to UART...
(14228) esp_core_dump_uart: Print core dump to uart...
(14234) esp_core_dump_common: Found tasks: (26)!
================= CORE DUMP START =================
WFMAAAEAAAAWAAAAcAEAAA==
FLX7P2Dn/D8M7vw/
```

sle118, 2020-08-27:
Could you please try the latest bleeding edge and see if the problem persists?

ElFishi, 2020-08-27:
moved to 103 and I am afraid, I can still (sometimes) crash the esp when changing sources. 
I may have a brownout issue I need to chase, though. 
Is there a reason to believe the esp draws more current when changing the streaming source?

sle118, 2020-08-27:
Ideally, it would be best to have a precise way of reproducing this; if this is codec related, then the type of file/stream becomes critical for a "successful failure".

sle118, 2020-08-27:
Adding @philippe44 to the conversation, in case he has some thoughts.

philippe44, 2020-08-27:
is this switching source or switching codecs?

ElFishi, 2020-08-28:
I'm on fw: development.108-I2S-4MFlash atm and I can not provoke a core dump.
I'll close this now and open another issue should I be able to produce reproducible errors.

ElFishi, 2020-08-28:
I still get core dumps, I just don't know how to trigger them. Attached a log file.
[core dump scarlet 200828 .txt](https://github.com/sle118/squeezelite-esp32/files/5143785/core.dump.scarlet.200828.txt)

sle118, 2020-08-28:
Your latest log allowed me to get the details of the core dump. This might help.

```
espcoredump.py v0.3-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x40095a5b       0x40095a5b <invoke_abort+27>
lbeg           0x40094e79       1074351737
lend           0x40094e89       1074351753
lcount         0xfffffffc       4294967292
sar            0x4      4
ps             0x60020  393248
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40095dc4       1074355652
a1             0x3ffce760       1073538912
a2             0x3ff480b8       1072988344
a3             0x2f     47
a4             0xc      12
a5             0xffffffff       -1
a6             0x0      0
a7             0xfffffffe       -2
a8             0x0      0
a9             0x3ffce740       1073538880
a10            0x0      0
a11            0x3ffce6f1       1073538801
a12            0x3ffce69f       1073538719
a13            0x31     49
a14            0x0      0
a15            0x3ffce6a2       1073538722

==================== CURRENT THREAD STACK =====================
#0  0x40095a5b in invoke_abort () at /var/lib/jenkins/workspace/v4.0/components/esp32/panic.c:157
#1  0x40095dc4 in abort () at /var/lib/jenkins/workspace/v4.0/components/esp32/panic.c:172
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

======================== THREADS INFO =========================
  Id   Target Id         Frame
* 1    <main task>       0x40095a5b in invoke_abort () at /var/lib/jenkins/workspace/v4.0/components/esp32/panic.c:157
  2    process 1         0x4023cf8e in esp_pm_impl_waiti () at /var/lib/jenkins/workspace/v4.0/components/esp32/pm_esp32.c:484
  3    process 2         0x4023cf8e in esp_pm_impl_waiti () at /var/lib/jenkins/workspace/v4.0/components/esp32/pm_esp32.c:484
  4    process 3         vTaskDelay (xTicksToDelay=<optimized out>) at /var/lib/jenkins/workspace/v4.0/components/freertos/tasks.c:1488
  5    process 4         0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffd794c, pvBuffer=0x3ffd8790, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  6    process 5         0x4009d217 in prvProcessTimerOrBlockTask (xListWasEmpty=<optimized out>, xNextExpireTime=38212) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  7    process 6         vTaskDelay (xTicksToDelay=<optimized out>) at /var/lib/jenkins/workspace/v4.0/components/freertos/tasks.c:1488
  8    process 7         0x4009ae08 in xQueueGenericReceive (xQueue=0x3f8137b0, pvBuffer=0x3ffd94f0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  9    process 8         0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffd75c4, pvBuffer=0x3ffcc260, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  10   process 9         0x4000bff0 in ?? ()
  11   process 10        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffe0c64, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  12   process 11        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffb7914, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  13   process 12        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffb7ed8, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  14   process 13        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffeaf00, pvBuffer=0x3ffebf30, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  15   process 14        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffe6d7c, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  16   process 15        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffe8270, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
  17   process 16        0x4009ae08 in xQueueGenericReceive (xQueue=0x3ffe9390, pvBuffer=0x0, xTicksToWait=<optimized out>, xJustPeeking=0) at /var/lib/jenkins/workspace/v4.0/components/freertos/include/freertos/portable.h:214
[... 41 lines cut ...]
.coredump.tasks.data 0x3ffb9e40 0x164 RW
.coredump.tasks.data 0x3ffb9cb0 0x188 RW
.coredump.tasks.data 0x3ffebfd4 0x164 RW
.coredump.tasks.data 0x3ffebe20 0x1ac RW
.coredump.tasks.data 0x3ffd62d4 0x164 RW
.coredump.tasks.data 0x3ffe7ae0 0x1a0 RW
.coredump.tasks.data 0x3ffd643c 0x164 RW
[... cut ...]
.coredump.tasks.data 0x3ffed340 0x1c0 RW
.coredump.tasks.data 0x3ffb1aec 0x164 RW
.coredump.tasks.data 0x3ffe6a90 0x1dc RW
.coredump.tasks.data 0x3ffdae4c 0x164 RW
.coredump.tasks.data 0x3ffdac80 0x1c4 RW
.coredump.tasks.data 0x3ffb758c 0x164 RW
.coredump.tasks.data 0x3ffb73f0 0x194 RW
.coredump.tasks.data 0x3ffbc7b8 0x164 RW
.coredump.tasks.data 0x3ffe5a50 0x218 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

sle118, 2020-08-28:
well...  looking at the core dump, the only thing I can see is that an assert failed while allocating a socket.  Stack isn't shown in these cases so it's hard to tell where this came from.

The 

```
alloc_socket(struct netconn *newconn, int accepted)
...
  /* The socket is not yet known to anyone, so no need to protect
         after having marked it as used. */
      SYS_ARCH_UNPROTECT(lev);
      sockets[i].lastdata.pbuf = NULL;
#if LWIP_SOCKET_SELECT || LWIP_SOCKET_POLL
      LWIP_ASSERT("sockets[i].select_waiting == 0", sockets[i].select_waiting == 0);
```
Seems like some code is creating a new a socket, and that doing so it finds a potential available slot to use. Then when checking that slot, the system finds out that the slot has the "select_waiting" flag set, which is an unexpected condition.  

If I were to make a guess here, it looks like a socket was being "waited on" and was freed incorrectly, leaving the select_waiting data dirty.

pgu-swir, 2020-08-28:
you are doing something very funny in this log, which is closing and opening a track in 400ms...

philippe44, 2020-08-28:
you are doing something very funny in this log, which is closing and opening a track within 400ms... but that put me probably on the right track. A general squeezelite problem though :-)

sle118, 2020-08-28:
More than likely an upstream fix to squeezelite, which could have been undetected on platforms with a different lwip/socket implementation. In our case, all the sockets memory is allocated upfront and reused. In other implementations, memory isn't an issue and is allocated at will.

philippe44, 2020-08-28:
> More than likely an upstream fix to squeezelite, which could have been undetected on platforms with a different lwip/socket implementation. In our case, all the sockets memory is allocated upfront and reused. In other implementations, memory isn't an issue and is allocated at will.
>
It's more that HL OS don't care if you poll a closed socket, they just ignore that. LWIP crashes. It is a rare case of race condition that is not easy to fix simply and I want small&simple

philippe44, 2020-08-29:
I've pushed a fix that I don't like but let me know if it works

ElFishi, 2020-08-29:
I may have switched rapidly between radio stations to make it trip. But I don't recall that I was the switching when the esp had the core dump.

sle118, 2020-08-29:
@ElFishi, the bleeding edge build now has a fix from @philippe44 : ce369638daf8a8c3308d28db6dd7b65123cd0161. Not sure if this is complete yet, but it's worth trying. Please report of this resolves the issue of not. The release to check for is: 111

ElFishi, 2020-08-29:
I ran 111 for most of the day w/o problems. In the log I find a few watchdog triggers: 
```
[03:59:10.290540] output_thread_i2s:473 Output state is 2
[03:59:11.290375] grfs_handler:738 scroller current size 1280 (w:228)
[03:59:11.298466] grfs_handler:738 scroller current size 1424 (w:228)
E (12528543) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12528543) task_wdt:  - IDLE1 (CPU 1)
E (12528543) task_wdt: Tasks currently running:
E (12528543) task_wdt: CPU 0: wifi
E (12528543) task_wdt: CPU 1: decode
[04:03:22.679288] stream_thread:327 end of stream
[04:03:35.433516] decode_thread:100 decode complete
[04:03:35.563535] codec_open:279 codec open: 'm'
[04:03:35.566854] stream_sock:495 connecting to 192.168.0.121:9000
[04:03:35.571943] stream_sock:572 header: GET /stream.mp3?player=30:ae:a4:c5:43:bc HTTP/1.0

[04:03:35.967342] stream_thread:254 headers: len: 491
HTTP/1.1 200 OK
```
or 
```
[05:05:02.706396] _output_frames:149 track start sample rate: 44100 replay_gain: 0
[05:05:02.709280] output_thread_i2s:473 Output state is 2
E (24123) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (24123) task_wdt:  - IDLE1 (CPU 1)
E (24123) task_wdt: Tasks currently running:
E (24123) task_wdt: CPU 0: IDLE0
E (24123) task_wdt: CPU 1: decode
[05:05:32.585066] grfs_handler:738 scroller current size 1280 (w:232)
[05:05:32.587637] grfs_handler:738 scroller current size 1440 (w:232)
[05:10:46.367602] stream_thread:327 end of stream
[05:10:59.216575] decode_thread:100 decode complete
[05:10:59.328896] codec_open:279 codec open: 'm'
[05:10:59.332250] stream_sock:495 connecting to 192.168.0.121:9000
[05:10:59.336961] stream_sock:572 header: GET /stream.mp3?player=30:ae:a4:c5:43:bc HTTP/1.0

[05:10:59.725132] stream_thread:254 headers: len: 491
HTTP/1.1 200 OK
```

I will test some more tomorrow

sle118, 2020-08-29:
Watch dog are ok. The device thread is a beast, especially when starting a new stream and building a buffer.

ElFishi, 2020-08-31:
I kept running the latest versions up to 115 the last days. fairly stable apart from WDTs.

sle118, 2020-09-01:
WDT are normal for some codecs; nothing to worry about.

I'm happy your problem is solved (so far). I'll go ahead and close the issue for now.

closed 2020-09-01 by sle118