sle118/squeezelite-esp32 #40: Radio Paradise flac stream crash

ElFishi, 2020-06-15:
The Radio Paradise flac stream crashes my Squeezelite-esp32 devices. This issue is to document one instance:

- hardware: WROVER + I2S DAC ES9023, , LEDs (see config below)
- firmware version used. [Philippe's fix version](https://forums.slimdevices.com/showthread.php?111328-ANNOUNCE-the-SqueezeAMP-next-generation-squeezeboxen-amplifier-receiver&p=978375&viewfull=1#post978375)
- display type:  SPI / SSD1322
- Rotary: yes
- Buttons: no
- IR receiver: yes
- Equalizer active: no
- Stats activated in NVS: yes
- Network encryption type: ?
- Are you syncing with other players: not this time, normally yes.
- LMS version: Logitech Media Server Version: 7.9.2 - 1578996832
- Squeezelite-ESP32 plugin version: 0.94
- Config.json: [Gru-config-json.txt](https://github.com/sle118/squeezelite-esp32/files/4779299/Gru-config-json.txt)
- Log: [Gru-crash-serial.log](https://github.com/sle118/squeezelite-esp32/files/4780050/Gru-crash-serial.log)

ElFishi, 2020-06-15:
Same System but 
- firmware version used: v0.5.597-IDFv3.1.5-I2S-4MFlash
- Log: [Gru-crash-serial-597.log](https://github.com/sle118/squeezelite-esp32/files/4780232/Gru-crash-serial-597.log)

ElFishi, 2020-06-15:
Same system
- firmware version used: v0.5.597-IDFv3.1.5-I2S-4MFlash
- display type: disabled
- Config.json: [Gru-config-json-no-screen.txt](https://github.com/sle118/squeezelite-esp32/files/4780366/Gru-config-json-no-screen.txt)
- Log: [Gru-crash-serial-597-noscreen.log](https://github.com/sle118/squeezelite-esp32/files/4780325/Gru-crash-serial-597-noscreen.log)

ElFishi, 2020-06-15:
Same system
- firmware version used: v0.5.597-IDFv3.1.5-I2S-4MFlash
- display type: disabled
- Rotary: disabled
- Config.json: [Gru-config-json-no-screen-norot.txt](https://github.com/sle118/squeezelite-esp32/files/4780514/Gru-config-json-no-screen-norot.txt)
- Log: [Gru-crash-serial-597-noscreen-norot.log](https://github.com/sle118/squeezelite-esp32/files/4780516/Gru-crash-serial-597-noscreen-norot.log)

ElFishi, 2020-06-15:
Second system
- hardware: WROVER + I2S DAC ES9018
- firmware version used: bleeding.edge.67-I2S-4MFlash
- display type: I2C/ SSD1306
- Rotary: no
- Buttons: no
- IR receiver: no
- Equalizer active: no
- Stats activated in NVS: yes
- Network encryption type: ?
- Are you syncing with other players: not this time, normally yes.
- LMS version: Logitech Media Server Version: 7.9.2 - 1578996832
- Squeezelite-ESP32 plugin version: 0.94
- Config.json: [Scarlet-config-json.txt](https://github.com/sle118/squeezelite-esp32/files/4782831/Scarlet-config-json.txt)
- Log: [Scarlet-crash.log](https://github.com/sle118/squeezelite-esp32/files/4782834/Scarlet-crash.log)

ElFishi, 2020-06-15:
Btw, I restarted LMS (which I had done before). 
Now the v0.5.597-system seems to run (longer). 
I am at 30 min...

ElFishi, 2020-06-16:
Btw, the radio paradise FLAC streams do not always stream flac. Sometime songs come as ogg. This may account for variations in the survival times. I found that by clicking on the song in LMS. The title information also contains the file type.

philippe44, 2020-06-18:
> * Log: [Scarlet-crash.log](https://github.com/sle118/squeezelite-esp32/files/4782834/Scarlet-crash.log)
>
That gives me a hint (not perfect) about something happening in the decoder, a buffer alignment issue or similar and the flac_decode method never consumes the decoded frames so decoder keeps knocking it relentlessly. I had a similar issue in pcm.

ElFishi, 2020-06-18:
Thx for picking this up. I am happy to try more ways to uncover additional hints if you tell me how.

ElFishi, 2020-08-14:
no crashes with fw: v0.5.641-IDFv3.1.5-I2S-4MFlash

closed 2020-08-14 by ElFishi

alidaf, 2024-02-07:
@ElFishi How did you get the ES9018 to work?

ElFishi, 2024-02-07:
What is your problem?

alidaf, 2024-02-08:
> What is your problem?

I'm just getting some low level music overpowered by really bad hiss in one speaker. I just dropped this board in place of a PCM5102 board. Did you ground the MCLK or just leave it unconnected? I've looked through your logs and I seem to have it set up exactly the same but wondered if you used some arcane magic because I can't figure it out.
There are 2 variants of the board and I'm wondering whether I was sent the reverse one, i.e. RCA input to I2S. The board says out on the RCA sockets so I'm probably clutching at straws.
I've tried resampling as well. Apparently the early problems on the RPi with this board were due to it being configured by default for 384k input.

ElFishi, 2024-02-09:
Yes, magic, thats me! 
Which board are you using? 
My ES9018 is a [Franken-Topping D10](https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/108002-announce-the-squeezeamp-next-generation-squeezeboxen-amplifier-receiver?p=1431176#post1431176): A stock D10 where I broke the PCB traces for DATA, LRCLK and BCLK and rerouted them to a ESP.  
So I can't really tell what MCLK is doing. 
To have it play 16bit you need to send the right command via I2C: 
`dac_controlset {"init":[ {"reg":1,"val":12} ],"poweron":[ {"reg":1,"val":12} ]}`

alidaf, 2024-02-09:
@ElFishi Wow, and oh dayum that's some serious L337 hackery (I am in awe). I'm at the much cheaper end of the market with one of [these](https://www.aliexpress.com/item/1005003740290057.html?spm=a2g0o.productlist.main.1.22819388EPQ3lU&algo_pvid=492d62a8-f05b-4d94-b2c8-edb722aa43f5&algo_exp_id=492d62a8-f05b-4d94-b2c8-edb722aa43f5-0&pdp_npi=4%40dis%21GBP%2120.13%2117.31%21%21%2124.67%2121.22%21%40211b441e17074681448106962ebe0e%2112000026997084921%21sea%21UK%214504794635%21&curPageLogUid=ZRaBnxaZWk9e&utparam-url=scene%3Asearch%7Cquery_from%3A), trying to play with the big boys. This board doesn't have an I2C connection so I think I'm stuck with whatever mode it's hard-wired in. It's really aimed at the RPi market where it seems it was as equally frustrating to get to work for a while but they now have an overlay that makes it work for them. Thanks for the input.

ElFishi, 2024-02-09:
I try my level best : ) 
the default input configuration of the ES9018K2M is I2S 32-bit. 
Have you tried a I2S-4MFlash-32 version of the FW?

alidaf, 2024-02-09:
> Have you tried a I2S-4MFlash-32 version of the FW?
I just beat you to the pip on that one. Quite literally moments ago. Instead of horrible noise, silence. I take that as progress so I'm just looking into the soxr parameters to see if there's anything I can do there. The board I have comes with the recommendation to use the hifiberry-dac overlay for the RPi, but the hifiberry-dac is based on a PCM5102 and I have one of those working well with the ESP32. Sigh, The ES9018 board has lots of stuff on it that looks like it all does something more than the PCM5102 and I'm a sucker for that. If it had anything that periodically went ping on it, I'd have written a blank cheque for it.

ElFishi, 2024-02-09:
Not sure I follow. I just switched to I2S-4MFlash.32.1670 and I use 
`"autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 300 -W  -n \"Scarlet\"",`
so no special resampling. 
What are "soxr parameters"?

alidaf, 2024-02-09:
Soxr is the linux code that Squeezelite-esp32 uses for resampling. It's activated using the -R switch in the autoexec1 parameters with various options. I think it's a dead end for this. Oh well, I'll put it on the back burner for now - plenty else to be getting on with. Thanks for the help though. Keep up the magic.

philippe44, 2024-02-09:
Soxr is not in squeezeesp32, I had to build another resampler as it was consuming way too much cpu

alidaf, 2024-02-09:
> Soxr is not in squeezeesp32, I had to build another resampler as it was consuming way too much cpu

Apologies. The guide says that the format is the same, hence the assumption.