sle118/squeezelite-esp32 #41: Disable led in software?

DarkShortie, 2020-06-15:
I've build a Squeezeamp and I'm using it in my sleeping room. When no playback is active the led is blinking which is relay annoying when you try to sleep.
Is it possible to disable the led completely with some software switch or nvs property?

ElFishi, 2020-06-16:
I don't have a SqueezeAMP, but I presume in the NVS config under set_GPIO it says 
12=green:0

If so, change it to 
12=Vcc
This will tell the system to keep GPIO12, which is connected to the green LED, high so it stays off.

DarkShortie, 2020-06-18:
set_GPIO was empty, I've changed it to 12=Vcc and rebooted, but nothing changed

ElFishi, 2020-06-18:
the other LED should be 13, but maybe someone who knows better should chime in...

philippe44, 2020-06-21:
It's fixed for SqueezeAMP, the set_GPIO only works if you use the I2S version. Having said that, you can use this version and set manually all SqueezeAMP I/O's except for the LEDs. I've verified that in the past when I moved all parameters to be usable from NVS

philippe44, 2020-07-29:
I've now added PWM control on the green and red LEDs. Use the NVS parameter 'led_brightness' and set a value between 0 and 100

closed 2020-07-29 by philippe44