sle118/squeezelite-esp32 #42: i2c is started. ESP_FAIL

ElFishi, 2020-06-24:
v0.5.601-IDFv3.1.5-I2S-4MFlash
that doesn't look OK?
```
i2cdetect
D (4392052) cmd_i2ctools: Initializing driver from configuration.
D (4392052) cmd_i2ctools: Loading configuration from nvs
D (4392052) cmd_i2ctools: Determining if i2c is started on port 1
D (4392062) cmd_i2ctools: i2c is started. ESP_FAIL
W (4392072) cmd_i2ctools: Stopping i2c driver on port 1
I (4392072) cmd_i2ctools: Initializing i2c driver configuration.
   mode = I2C_MODE_MASTER,
   scl_pullup_en = GPIO_PULLUP_ENABLE,
   i2c port = 1,
   sda_io_num = 22,
   sda_pullup_en = GPIO_PULLUP_ENABLE,
   scl_io_num = 23,
   scl_pullup_en = GPIO_PULLUP_ENABLE,
   master.clk_speed = 400000
E (4392082) i2c: /var/opt/esp-idf/components/driver/i2c.c:1273 (i2c_master_cmd_begin):i2c driver not installed
D (4392102) cmd_i2ctools: Installing i2c driver on port 1
E (4392112) I2CDefaultWriteBytes: i2c_master_cmd_begin( I2CPortNumber, CommandHandle, I2CWait ) != ESP_OK, result: 259
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

------------------------------------------------------------------------------------
Detected the following devices (names provided by https://i2cdevices.org/addresses).
60 [3ch]- SSD1305 SSD1306 PCF8578 PCF8569 SH1106
------------------------------------------------------------------------------------
```

sle118, 2020-06-25:
Please try the CMake branch (bleeding edge). I have implemented additional features and fixes to i2c command line there

ElFishi, 2020-06-28:
cmake works alright when called via the serial interface. for telnet see issue #43.

sle118, 2020-08-27:
I guess the only remaining issue now is with command line, so I'll go ahead and close this one

closed 2020-08-27 by sle118