sle118/squeezelite-esp32 #43: some commands not recognised in telnet

ElFishi, 2020-06-26:
hw: hardware: WROVER + I2S DAC
fw: bleeding.edge.69-I2S-4MFlash
config:
```
sent config JSON with data: {"timestamp":1593187540065,"config":{"a2dp_ctmt":{"value":"1000","type":33},"a2dp_ctrld":{"value":"500","type":33},"a2dp_dev_name":{"value":"Squeezelite","type":33},"a2dp_sink_name":{"value":"SMSL BT4.2","type":33},"actrls_config":{"value":"","type":33},"airplay_name":{"value":"ESP32-AirPlay-Scarlet","type":33},"airplay_port":{"value":"5000","type":33},"ap_channel":{"value":"1","type":33},"ap_ip_address":{"value":"192.168.4.1","type":33},"ap_ip_gateway":{"value":"192.168.4.1","type":33},"ap_ip_netmask":{"value":"255.255.255.0","type":33},"ap_pwd":{"value":"squeezelite","type":33},"ap_ssid":{"value":"squeezelite-Scarlet","type":33},"autoexec":{"value":"1","type":33},"autoexec1":{"value":"squeezelite -b 500:2000 -d all=info -n \"Scarlet\" -o I2S","type":33},"autoexec2":{"value":"i2cconfig -l","type":33},"autoexec3":{"value":"i2cset -c 72 -r 1 12","type":33},"bat_config":{"value":"","type":33},"bt_name":{"value":"ESP32-BT-Scarlet","type":33},"bt_sink_pin":{"value":"1234","type":33},"bypass_wm":{"value":"0","type":33},"dac_config":{"value":"bck=25,ws=33,do=32","type":33},"display_config":{"value":"I2C,width=128,height=64,address=60,driver=SSD1306","type":33},"enable_airplay":{"value":"1","type":33},"enable_bt_sink":{"value":"1","type":33},"host_name":{"value":"Scarlet","type":33},"i2c_config":{"value":"scl=13,sda=14,speed=400000,port=1","type":33},"jack_mutes_amp":{"value":"n","type":33},"lms_ctrls_raw":{"value":"n","type":33},"metadata_config":{"value":"","type":33},"ota_erase_blk":{"value":"249856","type":33},"ota_prio":{"value":"6","type":33},"ota_stack":{"value":"10240","type":33},"release_url":{"value":"https://github.com/sle118/squeezelite-esp32/releases","type":33},"rotary_config":{"value":"","type":33},"set_GPIO":{"value":"19=green:1,18=red:1,26=amp,15=ir","type":33},"spdif_config":{"value":"","type":33},"spi_config":{"value":"","type":33},"stats":{"value":"n","type":33},"telnet_block":{"value":"500","type":33},"telnet_buffer":{"value":"40000","type":33},"telnet_enable":{"value":"y","type":33}}}
```

I can issue commands to squeezelite-esp32 via the serial interface as expected
```
squeezelite-esp32> <\r><\n>
squeezelite-esp32> i2ccheck<\r><\n>
<\r><\n>
<27>[0;32mI (34417) messaging: i2c is started on port 1.<27>[0m<\r><\n>
squeezelite-esp32> <\r><\n>
squeezelite-esp32> free<\r><\n>
<\r><\n>
<27>[0;32mI (39657) messaging: 1413716<27>[0m<\r><\n>
squeezelite-esp32> <\r><\n>
squeezelite-esp32> heap<\r><\n>
<\r><\n>
<27>[0;32mI (44137) messaging: min heap size: 1393056<27>[0m<\r><\n>
squeezelite-esp32> <\r><\n>
squeezelite-esp32> i2cdetect<\r><\n>
<\r><\n>
<27>[0;32mI (49177) messaging: <\r><\n>
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f<\r><\r><\n>
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- <\r><\r><\n>
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- <\r><\r><\n>
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- <\r><\r><\n>
<\r><\r><\n>
------------------------------------------------------------------------------------<\r><\r><\n>
Detected the following devices (names provided by https://i2cdevices.org/addresses).<\r><\r><\n>
60 [3ch]- SSD1305 SSD1306 PCF8578 PCF8569 SH1106<\r><\r><\n>
72 [48h]- PCA9685 INA219 PN532 TMP102 INA260 ADS1115<\r><\r><\n>
------------------------------------------------------------------------------------<\r><\r><\n>
<27>[0m<\r><\n>
squeezelite-esp32> <\r><\n>
squeezelite-esp32> i2cdump -c 72<\r><\n>
<\r><\n>
<27>[0;32mI (59327) messaging: <\r><\n>
    00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f    0123456789abcdef<\r><\r><\n>
00: 00 0c 18 10 00 68 4a 80 10 22 05 02 5a 40 8a 00    .???.hJ??"??Z@?.<\r><\r><\n>
10: 00 ff ff ff 7f 00 00 00 00 00 00 00 00 00 00 XX    ....?..........X.<\r><\r><\n>
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.<\r><\r><\n>
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.<\r><\r><\n>
40: 30 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00    0?..............<\r><\r><\n>
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 60 00 19    .............`.?<\r><\r><\n>
60: 26 85 XX XX XX XX XX XX XX XX XX XX XX XX XX XX    &?X.X.X.X.X.X.X.X.X.X.X.X.X.X.<\r><\r><\n>
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.<\r><\r><\n>
<27>[0m<\r><\n>
squeezelite-esp32> <\r><\n>
```

If I try the same via telnet I get errors for some commands
```
help
E (13369) console: Unrecognized command: help

free
I (17569) messaging: 1389084
heap
E (23719) console: Unrecognized command: heap

i2cdetect
E (30669) console: Unrecognized command: i2cdetect

i2cdump -c 72
I (41269) messaging:
    00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f    0123456789abcdef
00: 00 0c 18 10 00 68 4a 80 10 22 05 02 5a 40 8a 00    .???.hJ??"??Z@?.
10: 00 ff ff ff 7f 00 00 00 00 00 00 00 00 00 00 XX    ....?..........X.
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.
40: 30 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00    0?..............
50: 00 00 00 00 00 00 00 00 08 00 00 00 00 00 00 19    ........?......?
60: 26 85 XX XX XX XX XX XX XX XX XX XX XX XX XX XX    &?X.X.X.X.X.X.X.X.X.X.X.X.X.X.
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.X.
```

sle118, 2020-06-27:
Which telnet client are you using? Does it happen systematically, or only under specific conditions? For example, command fails only the first time, then it works, etc.

ElFishi, 2020-06-28:
PuTTY 0.70
feels systematic to me. 
i2cdetect never seems to work, i2cdump -c 72 always works.
i2ccheck -p 1 works, but i2cconfig gives an interesting result
```
i2cconfig -l
E (225119) messaging: i2cconfig: invalid option "-?"
i2cconfig: invalid option "-?"
```

sle118, 2020-06-29:
This might have to do with the telnet client's line termination. I'll see if I can add an option to dump the incoming hex buffer to either confirm or rule that possibility out.

ElFishi, 2020-06-29:
I use hterm as telnet client for the serial interface. It allows me to choose the termination of input lines to be CR, LF or CR-LF. 
All of three options work on serial.

sle118, 2020-09-01:
@ElFishi  this should now be fixed, so I'll go ahead and close this issue

closed 2020-09-01 by sle118

ElFishi, 2020-09-01:
fw: development.121-I2S-4MFlash
```
help
E (26800) console: Unrecognized command: help
```

ElFishi, 2020-09-02:
telnet commands now work in fw: development.123-I2S-4MFlash.

"cursor up" "return" in telnet still crashes the system...