sle118/squeezelite-esp32 #45: some mp3 files play half(?) speed

daduke, 2020-08-05:
noticed this when playing my favorite internet radio station, so it's a bit convoluted to reproduce:

- install my wefunk LMS plugin https://daduke.org/linux/lmswefunk.html
- go to app -> Wefunk Radio -> play specific show -> 1038
- select song 8, Guajiro Bacan
- plays too slow

squeezelite on my Fedora workstation plays it just fine. All my squeezesp32 play it slow. Happens with selected songs from different shows on that station.

daduke, 2020-08-05:
provided example songs on https://daduke.org/junk/good.mp3 and bad.mp3.
can reproduce the problem on master and cmake branch, BT and DAC output

sle118, 2020-08-27:
@daduke does it behave the same way with the latest bleeding edge build?

philippe44, 2020-08-27:
ahah :-) only happens when streaming the file, hence I did not see it before. Not when playing local

daduke, 2020-08-27:
aha! the plot thickens!

philippe44, 2020-08-27:
In fact the file start with a wrong header, it's not properly mp3 aligned and that seems to confused mad when sent as a stream. As a file, LMS strips that bad header and so it sends something clean to the player

philippe44, 2020-08-27:
mistery solved:
The file is not aligned properly and start with a non-mp3 header (FFF). The mad decoder takes a fair bit of attempts before getting re-synchronized and he has one "false success" with a frame that tells sampling rate is 32k. The way squeezelite works is that at first successful frame it sets sampling rate, so 32k in that case
Then, further decode fail because the codec is still not aligned in reality, so it skips tills it reaches proper sync but it has kept a sampling rate of 32k and that cannot be changed
The funny & sad thing is that I tried with my AirPlay bridge and it worked so I scratched my head a lot, then I realized that AriPlay requires all sampling rate to be 44.1k. So when finding 32k, it starts resampling process but when later it finds propered sync'd frames, resampling process works just fine.
I've verified with my other bridges and the problem of 32/44.1 exists with all of them. Also tried with genuine squeezelite and that's the same!
So, no shame on SqueezeESP32, shame on your mp3 file ;-)

daduke, 2020-08-27:
thanks for the explanation. First off, it's not 'my' mp3 file, it's Wefunk's. Second, squeezelite on my Linux box plays it just fine, so there must be some ESP32 component to the problem.. close it if you think the the source's fault, I just wanted to report it because it happens on a number of individual songs in multiple wefunk shows.

philippe44, 2020-08-27:
come-on, I'm pulling your leg :-)

But I can guarantee you that this file does not play well on my squeezelite (old version) with a Pi, exact same symptom with same logic. I'll re-check if @ralphy has made some hardening of mad on squeezelite. You can re-compile squeezelite to log every sync error and you'll see the flood when starting that file

daduke, 2020-08-27:
> come-on, I'm pulling your leg :-)
yeah but I do feel bad for wasting your time when it's the file that's corrupt. I was thrown off by the fact that all my other (non-squeezesp32) players never had an issue..
I can also report the problem to Professor Groove if you're positive it's his files that are to blame, fix it at the source.

sle118, 2020-08-27:
Yay, we got traction!

philippe44, 2020-08-27:
Look at the file with a hex editor, there is something with is neither an ID3 nor a FFF audio header for the 1st couple of 100s bytes. I agree We could try to harden mad, but I still have to figure out how.

philippe44, 2020-08-28:
I've added an extra header check in mad which seems good enough - always listening to my friends :-)

daduke, 2020-08-28:
you're the best. I'll test it during the weekend.

sle118, 2020-08-28:
https://github.com/sle118/squeezelite-esp32/commit/7f671909bbcc37c53101e444de65b43f2c063cd0

daduke, 2020-08-28:
> I've added an extra header check in mad which seems good enough - always listening to my friends :-)

fixed! Awesome.

closed 2020-08-28 by daduke