sle118/squeezelite-esp32 #47: development.108-I2S-4MFlash loses WiFi

ElFishi, 2020-08-28:
My esp frequently loses the WiFi connection: 
Sound stops, display freezes, and web interface is not available.  
I can issue a restart command via the serial interface and the WiFi connects w/o problems. LMS shows the player with a signal stregth of 73%.

log file attached.
[wifi lost scarlet 200828.txt](https://github.com/sle118/squeezelite-esp32/files/5143933/wifi.lost.scarlet.200828.txt)

philippe44, 2020-08-29:
Can you try to reset your router? I did not change anything wrt WiFi recently and I've send once or twice the same thing happening to me and everytime it was because something went south in my router

kheperV3, 2020-08-29:
Maybe it's a SPIFFS problem (WiFi needs it...)
Try to reformat...

sle118, 2020-08-29:
@philippe44 we did change a config earlier this year, but it was to resolve an issue. Is it possible that new config options were added in idf V4?

philippe44, 2020-09-01:
Does it still happen?

ElFishi, 2020-09-01:
hasn't for a while now.

closed 2020-09-01 by ElFishi