sle118/squeezelite-esp32 #48: Update certificate for old recovery to work

philippe44, 2020-08-30:
All current SqueezeAMP have the initial recovery and are facing the certificate issue which means it's impossible to update them to any new version (cmake or make), unless a local webserver is used which is usually difficult for most users. Can you think of a way to have the certificate updated so that "old" recovery can go on GH?

daduke, 2020-08-30:
would it be an idea to set up a public HTTP server to distribute the migration upgrade once to users who need it? It's a question of trust, but in the end they have to trust us anyway....

sle118, 2020-08-30:
@daduke has a point. The beauty of the approach is that users would "pull" a new release down from that server, which would make it possible for anyone to update, including the older SqueezeAmp in the wild.

daduke, 2020-08-30:
I'd be happy to set up a vhost on my server

philippe44, 2020-08-30:
I don't have a preference, candidly - as long as users are not stuck on old version or don't have to do contorsions to get the new ones, I'm fine. Whatever is easiest for you. What we don't want is your server to be hacked and populated with a hacked firmware that would sniff people's network and do I-don-t-know-what

sle118, 2020-08-30:
All we need is at least one build of each target, preferably the development branch. No need to keep the builds up to date; this is just to allow a single update to certificates. Once the update is done, further updates can happen from GitHub.

Philippe could update the first post of each SqueezeAmp thread with the links and instructions.

daduke, 2020-08-31:
> I don't have a preference, candidly - as long as users are not stuck on old version or don't have to do contorsions to get the new ones, I'm fine. Whatever is easiest for you. What we don't want is [...]

well I would like to not have my server hacked for a whole bunch of other reasons...

my initial attempt of providing http://squeezeamp.daduke.org turned into a train wreck pretty fast as I have enabled HSTS for all my subdomains, hence https only. Do we want a cheap bespoke domain for this purpose? squeezeampupgrade.com?

philippe44, 2020-10-12:
I've not seen users complaining about this, so I'll close it

closed 2020-10-12 by philippe44