sle118/squeezelite-esp32 #55: Changing audio jack behavior via SqueezeESP32 Configuration page does not seem to work

CornelisJ, 2020-10-11:
Observed with: fw: development.148-SqueezeAmp4MBFl

After plugging in an audio jack in my SqueezeAMP, the audio amplifier of the TAS57xx IC stayed active. Parameter jack_mutes_amp in the NVS-editor was set to "n" (without the quotes).  Then I tried to change the audio jack behavior via Configuration => Audio => General Audio Options. I changed the setting to "Headphones", and then hit the Save and Apply buttons. But after reboot, parameter  jack_mutes_amp in the NVS-editor was set to "n". So changing this parameter via the Audio Configuration page does not seem to work.

I could only change the audio jack behavior by changing parameter  jack_mutes_amp from "n" to "y" in the NVS-editor.

sle118, 2020-10-12:
Thank you for reporting this. I will have a look

sle118, 2020-10-18:
The bug was identified, and a fix i is under way.

sle118, 2020-11-21:
this should now be resolved

closed 2020-11-21 by sle118