sle118/squeezelite-esp32 #58: Dockerfile broken

MatthiasLienhard, 2020-11-18:
Following the build instructions I ran into the following issue:

```
> docker build -t esp-idf .
[...]
Step 11/17 : RUN cp /workspace/squeezelite-esp32/idf-patch/i2s.c /workspace/esp-idf/components/driver/i2s.c
 ---> Running in e9de5f670c13
cp: cannot stat '/workspace/squeezelite-esp32/idf-patch/i2s.c': No such file or directory
```
Apparently this patch has been removed

sle118, 2020-11-22:
You are correct.  The i2s driver was actually integrated as a component to avoid issues during the builds. I'll have a look at the docker.

sle118, 2020-11-22:
Removed the patching command from docker in commit 99722a6f9414c1395ac6a2fc1bd2f6d6dce570fc

closed 2020-11-22 by sle118