sle118/squeezelite-esp32 #60: A1S 526/529 Unresponsive web ui after saving Bluetooth Sink name

KeBul, 2020-11-22:
On a freshly flashed or ota upgraded version 526 or 529 after entering and saving a Bluetooth Sink device name the web ui would become completely unresponsive.
Connecting to the device console using putty showed it to be in a crash and reboot loop... attached are the console outputs from a 526 occurrence and 529 occurrence.
[526failedstartafter BTSink Set.txt](https://github.com/sle118/squeezelite-esp32/files/5580374/526failedstartafter.BTSink.Set.txt)
[529failedstartafter BTSink Set.txt](https://github.com/sle118/squeezelite-esp32/files/5580375/529failedstartafter.BTSink.Set.txt)

and the NVS config txt file for I think prior to the 529 attempt
[nvs_config1606005485824json.txt](https://github.com/sle118/squeezelite-esp32/files/5580379/nvs_config1606005485824json.txt)

sle118, 2021-01-14:
A change was made to the bluetooth driver to address this: 7c13c130b8d991f725506d7def7155a48a3cf69b

You should try the latest release and report

KeBul, 2021-01-17:
Thanks, tested on version 552, issue is fixed for me.

Please close

Thanks

Kev

closed 2021-03-09 by sle118