sle118/squeezelite-esp32 #61: A1S 526/529 - Squeezelite start line missing Bluetooth Sink name

KeBul, 2020-11-22:
On A1S 526 or 529, when changing the Bluetooth sink name on the Audio + LMS web ui page the a2dp_sink_name parameter in NVS gets changed but the squeezelite startup line in autoexec1 loses -n 'new sinkname'" from the BT section.
Even the message in 529 web ui confirming the change/save show the missing name parameter.

Console output, NVS config text file upload and screenshot after changing bluetooth sink from SoundCore 2 to Bose OE SoundLink...

[BT Sink.txt](https://github.com/sle118/squeezelite-esp32/files/5580437/BT.Sink.txt)

[nvs_config1606045361050json.txt](https://github.com/sle118/squeezelite-esp32/files/5580409/nvs_config1606045361050json.txt)

[image]

sle118, 2021-01-14:
The logic for bluetooth a2dp_sink_name was modified in that release.  The bluetooth sink name option will no longer be send down from Squeezelite itself, as it is now solely retrieved from the nvs value for a2dp_sink_name. 

The sink name assignment also has been completely changed in the current development state of the UI, and a list of nearby playback devices along with their rssi level will be shown for users to select.  

Config area
[image]

Drop down populated
[image]

KeBul, 2021-01-17:
OK understand the change in the way it works and testing version 552 confirms the new method is good and that BT is correctly "BT" in the squeezelite command line.

I'm not seeing a drop down box of available devices in V552 on my A1S Audio Dev kit but that's OK 

Happy for this to be closed.