sle118/squeezelite-esp32 #62: A1S 526/529: wifi disconnects, slimproto_run 602 errors, BT_APPL underflow error causing skip to next track

KeBul, 2020-11-22:
Initially thought these were just occurring when connected to a Bluetooth sink but it also occurred when configured for I2S and using wired headphones plugged in the headphone jack.
Basically when playing through a playlist (local VBR MP3s), pretty much every track fails to complete, playback skipping to the next track at any any seemingly random time.
The console logs shows these are caused by wifi disconnects/slimproto disconnects and underflow errs - not an expert but would guess the wifi disconnect is the cause of the others 
For me this behaviour is unique to 526/529, I've used 142 for many hours at a time with no skips, but get this behaviour as soon as I upgrade to 526 or 529, upgrade can be a re-flash from scratch after flash erase using ESP32 download tool or OTA upgrade from v142
It makes 526/529 unusable for me.

If I OTA downgrade to 142, let it complete and reboot, no other changes, then playback is perfect with no wifi disconnects

A couple of console logs:

[wifidropsandbrokenplayandskips.txt](https://github.com/sle118/squeezelite-esp32/files/5580492/wifidropsandbrokenplayandskips.txt)

[wifilostskippingtracks2.txt](https://github.com/sle118/squeezelite-esp32/files/5580501/wifilostskippingtracks2.txt)

sle118, 2021-01-14:
We discovered that there was an issue with the ESP-IDF release used during the GitHub build process. It was updated starting on build number 546. Please test again and report if the situation improved.

philippe44, 2021-01-16:
Can we close this one?

KeBul, 2021-01-17:
Hi Guys,

I've been testing on Version 552 and can confirm I'm no longer seeing this issue.

Thanks for your time and effort to resolve, I'm happy for you to close this now.

closed 2021-01-23 by sle118