sle118/squeezelite-esp32 #63: Compiling error esp-idf/components/driver/i2s.c:66:5: error: unknown type name 'SemaphoreHandle_t'

RamShadow, 2020-11-24:
Hey,
trying build from src. 
Debian WSL and Windows directly ends with:

Can anyone give me a hint or tell me that the error is real.

PS C:\Users\RamShadow\repo\SqueezeElite\squeezelite-esp32> idf.py all
Executing action: all (aliases: build)
Running cmake in directory c:\users\ramshadow\repo\squeezeelite\squeezelite-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\users\ramshadow\repo\squeezeelite\squeezelite-esp32"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.29.2.windows.2")
-- Building RECOVERY
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/RamShadow/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/RamShadow/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/RamShadow/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/RamShadow/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/RamShadow/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: custom.build
-- Building ESP-IDF components for target esp32
warning: LOGGING_SLIMPROTO (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:3) has leading or trailing whitespace in its prompt
warning: LOGGING_STREAM (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:8) has leading or trailing whitespace in its prompt
warning: I2S_NUM (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:93) has leading or trailing whitespace in its prompt
warning: I2S_BCK_IO (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:98) has leading or trailing whitespace in its prompt
warning: I2S_WS_IO (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:103) has leading or trailing whitespace in its prompt
warning: I2S_DO_IO (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:108) has leading or trailing whitespace in its prompt
warning: I2S_DI_IO (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:113) has leading or trailing whitespace in its prompt
warning: I2C_SDA (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:120) has leading or trailing whitespace in its prompt
warning: I2C_SCL (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:125) has leading or trailing whitespace in its prompt
warning: A2DP_CONTROL_DELAY_MS (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:186) has leading or trailing whitespace in its prompt
warning: style: quotes recommended around default value for string symbol AIRPLAY_PORT (defined at C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main/Kconfig.projbuild:224)
Loading defaults file C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/sdkconfig.defaults...
-- Found PythonInterp: C:/Users/RamShadow/.espressif/python_env/idf4.1_py3.9_env/Scripts/python.exe (found version "3.9")
-- Found Perl: C:/Perl64/bin/perl.exe (found version "5.24.3")
-- Adding linker script C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/RamShadow/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Components: app_recovery app_squeezelite app_trace app_update asio audio bootloader bootloader_support bt cbor coap codecs console cxx display driver driver_bt efuse esp-dsp esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon platform_config platform_console protobuf-c protocomm pthread raop sdmmc services soc spi_flash spiffs squeezelite squeezelite-ota tcp_transport tcpip_adapter telnet tools ulp unity vfs wear_levelling wifi-manager wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/platform_console/app_recovery C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/platform_console/app_squeezelite C:/Users/RamShadow/esp-idf/components/app_trace C:/Users/RamShadow/esp-idf/components/app_update C:/Users/RamShadow/esp-idf/components/asio C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/audio C:/Users/RamShadow/esp-idf/components/bootloader C:/Users/RamShadow/esp-idf/components/bootloader_support C:/Users/RamShadow/esp-idf/components/bt C:/Users/RamShadow/esp-idf/components/cbor C:/Users/RamShadow/esp-idf/components/coap C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/codecs C:/Users/RamShadow/esp-idf/components/console C:/Users/RamShadow/esp-idf/components/cxx C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/display C:/Users/RamShadow/esp-idf/components/driver C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/driver_bt C:/Users/RamShadow/esp-idf/components/efuse C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/components/esp-dsp C:/Users/RamShadow/esp-idf/components/esp-tls C:/Users/RamShadow/esp-idf/components/esp32 C:/Users/RamShadow/esp-idf/components/esp_adc_cal C:/Users/RamShadow/esp-idf/components/esp_common C:/Users/RamShadow/esp-idf/components/esp_eth C:/Users/RamShadow/esp-idf/components/esp_event C:/Users/RamShadow/esp-idf/components/esp_gdbstub C:/Users/RamShadow/esp-idf/components/esp_http_client C:/Users/RamShadow/esp-idf/components/esp_http_server C:/Users/RamShadow/esp-idf/components/esp_https_ota C:/Users/RamShadow/esp-idf/components/esp_https_server C:/Users/RamShadow/esp-idf/components/esp_local_ctrl C:/Users/RamShadow/esp-idf/components/esp_netif C:/Users/RamShadow/esp-idf/components/esp_ringbuf C:/Users/RamShadow/esp-idf/components/esp_rom C:/Users/RamShadow/esp-idf/components/esp_serial_slave_link C:/Users/RamShadow/esp-idf/components/esp_websocket_client C:/Users/RamShadow/esp-idf/components/esp_wifi C:/Users/RamShadow/esp-idf/components/espcoredump C:/Users/RamShadow/esp-idf/components/esptool_py C:/Users/RamShadow/esp-idf/components/expat C:/Users/RamShadow/esp-idf/components/fatfs C:/Users/RamShadow/esp-idf/components/freemodbus C:/Users/RamShadow/esp-idf/components/freertos C:/Users/RamShadow/esp-idf/components/heap C:/Users/RamShadow/esp-idf/components/idf_test C:/Users/RamShadow/esp-idf/components/jsmn C:/Users/RamShadow/esp-idf/components/json C:/Users/RamShadow/esp-idf/components/libsodium C:/Users/RamShadow/esp-idf/components/log C:/Users/RamShadow/esp-idf/components/lwip C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/main C:/Users/RamShadow/esp-idf/components/mbedtls C:/Users/RamShadow/esp-idf/components/mdns C:/Users/RamShadow/e [... cut ...] roject/main C:/Users/RamShadow/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/RamShadow/esp-idf/components/partition_table C:/Users/RamShadow/esp-idf/components/soc C:/Users/RamShadow/esp-idf/components/spi_flash C:/Users/RamShadow/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/RamShadow/repo/SqueezeElite/squeezelite-esp32/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

RamShadow, 2020-11-24:
esp-idf was new installed in WSL and in Windows direct. 
esp-idf 4.1
python 3.9
Perl: 5.24.3

shaffenmeister, 2020-11-26:
Please check with esp-idf 4.0 which is (AFAIK) the targeted framework. I can compile it on both Windows and Linux flawlessly with esp-idf 4.0.

RamShadow, 2020-11-26:
Thanks a lot.
4.0 Works!
(was my the latest and greatest paranoia...)

closed 2020-11-26 by RamShadow