sle118/squeezelite-esp32 #65: Ad Snapclient as Feature

AxelJust, 2020-12-20:
Hello everyone,
before I found this today, I looked at [snapclient by jorgenkraghjakobsen](https://github.com/jorgenkraghjakobsen/snapclient).
Do you think it would be posible to ad the Snapclient Feature to the Squeezelite32-esp32 Project?
I would also love to see the Spotify-Conect Feature Server Independant on the Esp.
Greatings
Axel Just

philippe44, 2021-01-08:
It would be a totally different animal then. Squeezelite-esp32 is made to be a LMS client in mulitroom and optionally a BT sink or an AirPlay( multi-sync) sync, but these last 2 are more unusual use. We might add a spotify client, but I'm not convince it will fit. I have spend a lot of effort & pain making that audio backend and the display control and it's very stable now. I'm hesitating to add something totally different, unless it's an independant application that cloud (e.g.) only fit side-by-side in the larger flash (up to 16MB) devices. Today, we need 4MB

sle118, 2021-12-30:
Given how little room we have left in Flash, and given we know support Spotify (albeit without sync), I'm going to close this as it won't be implemented.

closed 2021-12-30 by sle118

gopromaster, 2022-01-21:
How can Spotify be used? It is listed as an option in the readme, and your answer suggests it, but I can't find any info on how to set it up/use it...