sle118/squeezelite-esp32 #66: A1S_542: You have to select Bit Depth without the need of it.

Harry-1999, 2021-01-06:
If you save a non SSD1326-display, the hardware/display-section asks you to set the Bit Depth.
You only should set this, if you realy want to use a SSD1326-display.

(my) saved result in NVS-Editor: (with selected "1" for BitDepth, otherwise i can't save it)
SPI:width=128,height=64,cs=12,reset=22,speed=8000000,driver=SSD1306,HFlip,VFlip

If you need the 1 even with other displays than SSD1326, then the default should be "1", not "---".

sle118, 2021-01-09:
The UI frontend is very generic and doesn't perform validation outside of predefined values for drop down, this is a backend change.

The system should actually allow '--' as a valid value that is not saved, only for drivers other than SSD1326.

Harry-1999, 2021-01-09:
[image]

sle118, 2021-01-09:
The system "should" but doesn't, thus this problem. ;)

sle118, 2021-01-14:
This is now resolved on my development system.

sle118, 2021-01-14:
Note:  since the parameter was defined in the backend as integer, the issue was addressed on the front end.  There was a bug in the command line construction logic which incorrectly included drop down default values. The logic was updated to generate a command line that only contains valid values.

sle118, 2021-04-18:
Please test the latest release and reopen of necessary. https://github.com/sle118/squeezelite-esp32/commit/5120029643f5f4fa8f64a542e4a5e0ef68859c0d should fix the issue

closed 2021-04-18 by sle118