sle118/squeezelite-esp32 #73: idf.py menuconfig not working with latest esp-idf

fc92, 2021-01-19:
I tried to compile without success.

1. with docker, on master branch: without any change in menuconfig
```
 git show
commit 3c418097b883d017055245f1e6a7943387f1814c (HEAD -> master, origin/master)
Author: Philippe G <philippe44@users.noreply.github.com>
Date:   Thu Dec 3 20:59:06 2020 -0800

    plugin: sync players artwork & spectrum typo

make defconfig
make menuconfig
make          
WARNING: esp-idf git submodule components/asio/asio may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/bt/controller/lib may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/cbor/tinycbor may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/json/cJSON may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/lwip/lwip may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mbedtls/mbedtls may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mqtt/esp-mqtt may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/nghttp/nghttp2 may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
Toolchain path: /workspace/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /workspace/esp-idf/requirements.txt are satisfied.
GENCONFIG
App "squeezelite" version: custom.build
WARNING: esp-idf git submodule components/asio/asio may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/bt/controller/lib may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/cbor/tinycbor may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/json/cJSON may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/lwip/lwip may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mbedtls/mbedtls may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mqtt/esp-mqtt may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/nghttp/nghttp2 may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: Missing submodule components/esp_wifi/lib_esp32...
Attempting 'git submodule update --init components/esp_wifi/lib_esp32' in esp-idf root directory...
Submodule 'components/esp_wifi/lib_esp32' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib_esp32'
Cloning into '/workspace/esp-idf/components/esp_wifi/lib_esp32'...
Submodule path 'components/esp_wifi/lib_esp32': checked out 'fbec234429466dc25291cd89ff6f94c6f7a525f0'
CC build/bootloader/bootloader_support/src/bootloader_flash.o
....
CC build/cbor/tinycbor/src/open_memstream.o
/workspace/esp-idf/components/cbor/tinycbor/src/open_memstream.c:45:4: error: #error "Cannot implement open_memstream!"
 #  error "Cannot implement open_memstream!"
    ^
/workspace/esp-idf/components/cbor/tinycbor/src/open_memstream.c:57:8: error: unknown type name 'RetType'
 static RetType write_to_buffer(void *cookie, const char *data, LenType len)
        ^
/workspace/esp-idf/components/cbor/tinycbor/src/open_memstream.c:57:64: error: unknown type name 'LenType'
 static RetType write_to_buffer(void *cookie, const char *data, LenType len)
                                                                ^
/workspace/esp-idf/components/cbor/tinycbor/src/open_memstream.c:82:12: warning: 'close_buffer' defined but not used [-Wunused-function]
 static int close_buffer(void *cookie)
            ^
[... 71 lines cut ...]
-- Adding linker script /home/f/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/f/esp/squeezelite-esp32/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/f/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/f/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_recovery app_squeezelite app_trace app_update asio audio bootloader bootloader_support bt cbor cmock coap codecs console cxx display driver driver_bt efuse esp-dsp esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon platform_config platform_console protobuf-c protocomm pthread raop sdmmc services soc spi_flash spiffs squeezelite squeezelite-ota tcp_transport tcpip_adapter telnet tinyusb tools ulp unity vfs wear_levelling wifi-manager wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/f/esp/squeezelite-esp32/components/platform_console/app_recovery /home/f/esp/squeezelite-esp32/components/platform_console/app_squeezelite /home/f/esp/esp-idf/components/app_trace /home/f/esp/esp-idf/components/app_update /home/f/esp/esp-idf/components/asio /home/f/esp/squeezelite-esp32/components/audio /home/f/esp/esp-idf/components/bootloader /home/f/esp/esp-idf/components/bootloader_support /home/f/esp/esp-idf/components/bt /home/f/esp/esp-idf/components/cbor /home/f/esp/esp-idf/components/cmock /home/f/esp/esp-idf/components/coap /home/f/esp/squeezelite-esp32/components/codecs /home/f/esp/esp-idf/components/console /home/f/esp/esp-idf/components/cxx /home/f/esp/squeezelite-esp32/components/display /home/f/esp/esp-idf/components/driver /home/f/esp/squeezelite-esp32/components/driver_bt /home/f/esp/esp-idf/components/efuse /home/f/esp/squeezelite-esp32/components/esp-dsp /home/f/esp/esp-idf/components/esp-tls /home/f/esp/esp-idf/components/esp32 /home/f/esp/esp-idf/components/esp_adc_cal /home/f/esp/esp-idf/components/esp_common /home/f/esp/esp-idf/components/esp_eth /home/f/esp/esp-idf/components/esp_event /home/f/esp/esp-idf/components/esp_gdbstub /home/f/esp/esp-idf/components/esp_hid /home/f/esp/esp-idf/components/esp_http_client /home/f/esp/esp-idf/components/esp_http_server /home/f/esp/esp-idf/components/esp_https_ota /home/f/esp/esp-idf/components/esp_https_server /home/f/esp/esp-idf/components/esp_hw_support /home/f/esp/esp-idf/components/esp_ipc /home/f/esp/esp-idf/components/esp_local_ctrl /home/f/esp/esp-idf/components/esp_netif /home/f/esp/esp-idf/components/esp_pm /home/f/esp/esp-idf/components/esp_ringbuf /home/f/esp/esp-idf/components/esp_rom /home/f/esp/esp-idf/components/esp_serial_slave_link /home/f/esp/esp-idf/components/esp_system /home/f/esp/esp-idf/components/esp_timer / [... cut ...]
  squeezelite.cmake:124 (___register_flash)
  CMakeLists.txt:8 (include)

-- Configuring incomplete, errors occurred!
See also "/home/f/esp/squeezelite-esp32/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
/usr/bin/cmake -S/home/f/esp/squeezelite-esp32 -B/home/f/esp/squeezelite-esp32/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```

Please advise on the esp-idf 4 version to use. My goal is to use A2DP source audio output on a naked ESP32 Wrover module for the moment.

shaffenmeister, 2021-01-20:
Hi,
as long as you are on branch release/v4.0 or release tags 4.0.x, everything should work. I am using release/4.0.
idf.py menuconfig works.

fc92, 2021-01-20:
Thanks a lot! v4.0.2 compiles for me!

closed 2021-01-20 by fc92