sle118/squeezelite-esp32 #74: Endless "switching amplifier ON" messages

troffasky, 2021-01-23:
ESP32 Audio Kit V2.2
fw: ESP32-A1S.552.master-cmake,

With "set_GPIO" set to "21=amp,39=jack:0" and the headphone jack empty, the following is printed every few seconds, forever:
```
[02:11:28.099475] jack_handler:149 switching amplifier ON
[02:11:32.099472] jack_handler:149 switching amplifier ON
[02:11:35.269468] jack_handler:149 switching amplifier ON
[02:11:43.249476] jack_handler:149 switching amplifier ON

```
For what it's worth, the amp appears to be powered continuously, I cannot hear any fluctuation in audio levels through the speakers when it is supposedly being switched on.
The GPIO numbers are correct, if I do not have this set then the amp is not powered at all when the headphone jack is empty.

philippe44, 2021-02-12:
GPIO 36 and 39 are input only an require pull-up otherwise they will fluctuate.

troffasky, 2021-02-12:
So is this a defect in how "ESP32 Audio Kit V2.2" has been designed?

philippe44, 2021-02-12:
It's more an esp32 issue

closed 2021-03-24 by philippe44