sle118/squeezelite-esp32 #75: no Autoplay // and which commands are possible from console

dkotte, 2021-01-24:
Hi all,
first: great project!
At LMS i can see the squeezelite-esp32 connected and last (radio-)stream is running again.
But i have no audio output.
I have to select "next" button within LMS or FHEM (or stop/play) to hear the radio stream.
Environment:
squeezelite-esp32 - current version without any phsical devices (button/rotary-encoder/display)
BT-Speaker - speaker is connecting if i power up the ESP32
Other sqeeze... devices (squeezelite at raspi/app at anroid) start playing after starting device or app.

is this behavior by design or am I doing something wrong?
Are there any parameters to control this?
I am happy to post serial / debug outputs.

Which commands can be used via the console?
I get:
13: 21: 53.332 -> (15672100) console: Unrecognized command: play 

Thanks, Dirk

daduke, 2021-01-29:
I've never seen this with any of my 5 squeezelite-esp32's (both I2S and BT). Instant playback on power on.

sle118, 2021-01-29:
There's a configuration option under the audio preference *in the server config* which determines that behavior.

dkotte, 2021-01-30:
Hi, Thanks.
i tried it with BT and different audio sources.
Without success. I have to jump within the title or restart the title or go to the next to start playing.
At LMS i see the title playing and the time running.
I connect a PCM5102 and run into the same problems.
At serial console i can't see the HTTP GET from ESP
If i jump within the title i get the following:
squeezelite-esp32>
squeezelite-esp32>
squeezelite-esp32> [00:00:25.054388] decode_flush:251 decode flush
[00:00:25.055333] output_flush:440 flush output buffer
[00:00:25.101945] codec_open:279 codec open: 'm'
[00:00:25.104113] stream_sock:508 connecting to 192.168.10.11:9000
[00:00:25.109670] stream_sock:585 header: GET /stream.mp3?player=7c:9e:bd:fb:e1:78 HTTP/1.0
[00:00:25.319964] stream_thread:264 headers: len: 116
HTTP/1.1 200 OK
Server: Logitech Media Server (7.9.2 - 1572699180)
Connection: close
Content-Type: audio/mpeg
[00:00:26.070991] mad_decode:247 setting track_start

in addition, I can hear the data, but not the sound (as in old times when loading the programs from diskette)

I have a VS1053 too. But i don't know how to connect.
Is it possible to use this device ... and which connections and settings I have to make.
it would be great if someone could help with this too.
Dirk

sle118, 2021-02-18:
As for the VS1053, you can't use it with this project.  You will need something based on a simple DAC like the PCM5102A (widely available and inexpensive), which is able to process an I2S data stream. 

The released firmware has limited log verbosity to reduce the binary size (debug strings take up a good amount of space!).  You can increase the Squeezelite's logs level to see if you find anything.

sle118, 2022-05-23:
Closing because of inactivity.

closed 2022-05-23 by sle118