sle118/squeezelite-esp32 #76: build problems

ole348, 2021-02-06:
fresh install of ubuntu 20.04.2
fresh install of ESP-IDF v4.3-dev-2940-g6e776946d
clone of cmake-master

can anybody please try this and maybe solve numerous build issues and missing files

shaffenmeister, 2021-02-06:
This projects aims at the ESP IDF 4.0 branch. To my knowledge newer releases do not work yet.

sle118, 2021-02-07:
Choosing the right ESP-IDF is critical to a stable build. We've seen several problems from various commits which caused instabilities to happen and as such, it is best to aligh with the officially supported version. An alternative method to build the project is to use docker, which is at the core of the existing automated build on GitHub. For example, the following should work for the I2S-4MFlash target:
`docker run --rm -v $PWD:/project -w /project sle118/idf:release-v4.0 /bin/bash -c "cp build-scripts/I2S-4MFlash-sdkconfig.defaults sdkconfig && idf.py build "`

note that this is based on a "cloned" docker instance from espressif, which is frozen as of the most stable release we found.

closed 2021-02-07 by sle118