sle118/squeezelite-esp32 #79: Preset-Buttons don't work... solution found

Mum-Pf, 2021-02-12:
Hi
the new preset-buttons from PR#64 have a litle typo-bug... you forgot a "_"-sign (underscore).
File components/squeezelite/controls.c
Line 145 – 150
old: LMS_CALLBACK(pre1, PRESET_1, preset1.single)
new : LMS_CALLBACK(pre1, PRESET_1, preset_1.single)
...
old: LMS_CALLBACK(pre6, PRESET_6, preset6.single)
new: LMS_CALLBACK(pre6, PRESET_6, preset_6.single)

I compiled the new lines and it worked :-)
Can you integrate it to your next version?

Many greetings and thank you
M.P.

sle118, 2021-02-13:
Please check and close the issue if it works

Harry-1999, 2021-02-13:
I tested ist... and now it works... 570:A1S
Presets and Power

Mum-Pf, 2021-02-13:
thanks, it works with v570 (A1S)

closed 2021-02-13 by Mum-Pf