sle118/squeezelite-esp32 #80: Volume Control: LMS volume and bluetooth sink volume are interlinked

shaffenmeister, 2021-03-04:
### Observation
When using squeezelite-esp32 as bluetooth speaker (bluetooth sink), the maximum possible volume depends on the last volume set via LMS.

This may result in a very low (or high) max volume when switching from LMS to bluetooth speaker mode or vice versa.

The (suspected) reason for this behaviour is that bluetooth and squeezelite volume control overlap.
It is not an issue when powering up without connecting to LMS.

### Expected behaviour
Not clear.
Possible scenarios to consider:
1. Switch from LMS to bluetooth speaker mode (bt sink).
2. Switch from bluetooth (bt sink) to LMS.

Possible issues:
* When switching between the above scenarios while playing audio, there may be sudden changes in volume which may be unwanted behaviour.

### EUT
SqueezeAmp.572.master-cmake

philippe44, 2021-03-07:
Your BT controller should send the volume every time it takes control of a device

shaffenmeister, 2021-03-07:
My BT controllers seem set the volume accordingly (tested with Linux, Android).

Example to clarify:
If I listen to music via LMS and set the "LMS volume" to 20 % and switch over to BT sink mode, I can set the "BT volume" 0 to 100% but the maximum "physical output volume" is limited to the "LMS volume" 20 %.

Still being in BT sink mode, I can set the squeezeamp "LMS volume" to 100 % via LMS which then provides me a "pyhiscal output volume" over the full physical range.

philippe44, 2021-03-07:
Oh really? That's very strange, I've never observed that. You are saying that the LMS volume set the maximum volume for BT. I'll re-check but nothing in the code seems to support that. 
Which DAC are you using and what is your squeezelite command line?

shaffenmeister, 2021-03-08:
I'll try to check this though I am not an expert in this regard. Will check it until the coming weekend. Need to do some reading first... ;-)

I can build my own image and would be happy to assist.

philippe44, 2021-03-21:
Did you try recent versions?

shaffenmeister, 2021-03-22:
Yes, unfortunately, my SqueezeAmp unit left "bricked" (error while flashing) and I did not find the time to test this in depth
as I was not able to flash my device again.

Before that I noticed some effects of your modifications but with my Android phone I ran into clipping issues.
I suppose the safest solution (if possible) could be to make both BT and squeezelite control the same "hardware" volume control
and neglect BT audio stream manipulations by the BT source.

sle118, 2021-03-25:
@shaffenmeister SqueezeAmp cannot really be bricked. If you want to discuss this, we sometimes hang around in gitter:

https://gitter.im/SqueezeAmp/community

sm6uax, 2021-04-05:
Hi

Having the same problem as well. Have never been able to compile it all so running from the updates.

The current fw is dev.634-32

Anyway many many thanks for the effort in the project. Currently I have 3 devices running but more components on the way

Br 

Per

sle118, 2021-04-14:
For the issue reported in this thread, please consider testing the fix introduced in 879b1f9107bbd99344f3ce2713f444afd1b27cc3

sm6uax, 2021-04-14:
Thanks alot.

From my initial test it seems to work very well. I will make some long time test. There is still a problem when switching between LMS and bluetooth but it will try to get an exact problem description in a new ticket.

Many thanks
Per

shaffenmeister, 2021-04-15:
For me, switching between LMS and BT (Android, Ubuntu) seems to work ok.
I.e. LMS volume and BT volume are now handled independently.
One has to reconnect via BT each time after audio playback via LMS.

sle118, 2021-04-18:
For now I will close this issue then.

closed 2021-04-18 by sle118