sle118/squeezelite-esp32 #82: Player name is not saved permanent

networxnet, 2021-03-26:
Player name is not saved permanent after change.
Name will reset to squeezelite-fbeXXX after every reboot.

networxnet, 2021-03-27:
Same prob with latest firmware, LMS 8.2.0

shaffenmeister, 2021-03-28:
### Where do you change the player's name?
1. On the player's own web UI --> Changing the name does work for me.
2. Changing the name via LMS --> Name is lost after reboot, as per your observation

### Observed behaviour on SqueezeliteESP32
Serial output after issuing the "change name command" via LMS web UI:
`[00:42:08.033724] process_setd:480 set name: SqueezeESP32b`

I.e. the command is sent to the client correctly, the client does not issue an error message.

### Environment:
* Logitech Media Server Version: 8.1.2 - 1616822325 @ Sat Mar 27 06:39:42 CET 2021
* fw: SqueezeAmp.16.629.master-cmake, mode: squeezelite

networxnet, 2021-03-28:
Changing from LMS ist not permanent (why?).

Changing from Players UI will have no effect on LMS.

[image]
[image]

networxnet, 2021-03-28:
After rewrite 629-16 firmware and change name from players ui to another value, save and change Back, it seems to work.

Open question: Why it does not work from LMS like other players?

philippe44, 2021-03-31:
It's because the SETD command is not honored by squeezelite-esp32. I would have to add modifications to the "stock" squeezelite. This is not much more a "stock" version now, so I'll do that in the coming days.

philippe44, 2021-04-01:
Done

closed 2021-04-01 by philippe44