sle118/squeezelite-esp32 #83: adc_power_always_on() not available in recent esp-idf 4.0 version (branch: release/4.0)

shaffenmeister, 2021-03-29:
**Observation**
In the past I could compile squeezelite-esp32 without issues using IDF branch release/4.0.
After fetching the latest IDF commits (until 41efdb0b34864d434ce144e8530a4a78dfd7cd9f), compilation fails:

> ../components/services/i2s.c: In function 'i2s_param_config':
> ../components/services/i2s.c:883:9: error: implicit declaration of function 'adc_power_always_on'; did you mean 'adc_power_release'? [-Werror=implicit-function-declaration]
>          adc_power_always_on();
>          ^~~~~~~~~~~~~~~~~~~
>          adc_power_release
> cc1: some warnings being treated as errors

**Expected behaviour**
Not clear, I think the API changed so that one needs to use adc_power_acquire() and adc_power_release().

The local squeezelite-esp32 i2s driver refers to that function which is not present in esp idf anymore.

sle118, 2021-03-30:
We have settled on a very specific version of the esp-idf that was providing a stable build and given the extensive amount of work required for regression testing, we have no plans to migrate the code to a different version in the short term.

Please make sure you are using the recommended ESP idf commit, or alternatively use the following Docker image: 
https://hub.docker.com/r/sle118/idf

closed 2021-03-30 by sle118

shaffenmeister, 2021-03-30:
Side note: The README does not state that specific version yet. I am going with the docker container, though.

besi, 2022-05-01:
@sle118 I'd prefere to use a git repository instead of a docker container. In a gitter-chat-log I did find the commit 28f1cdf5ed7149d146ad5019c265c8bc3bfa2ac9 of the ESP-IDF. However I still get the error:

    squeezelite/components/services/i2s.c:883:9: error: implicit declaration of function 'adc_power_always_on'; did you mean 'adc_power_release'? [-Werror=implicit-function-declaration]

Could you maybe specify which commit is being used to compile the current version of squeezelite?