sle118/squeezelite-esp32 #91: Crackle when using Airplay

fieldOfView, 2021-04-23:
When streaming audio via to squeezelite-esp32 over Airplay, I get a lot of crackle over the audio. Streaming the same source over Bluetooth works well.

I am using an ESP32-A1S, and tried self-compiled builds as well as the 1.665 build. After reading #35 I also tried dev.552, all with the same crackling result.

Are there any things I could try to get rid of the crackle?

philippe44, 2021-06-30:
What is your audio controller (airplay)

fieldOfView, 2021-06-30:
I tried my iPhone (ios 14.4) and a Synology NAS (AudioStation application).

fieldOfView, 2021-07-15:
It turns out that using my iPhone as the audio controller does not exhibit the crackling issue. I will see if I can reproduce it it other unofficial Airplay sources.

fieldOfView, 2021-07-15:
Looking at the idf monitor output I can see there are a lot of dropped frames when playing from the Synology NAS.

[monitor output of ~15 seconds played through Synology NAS](https://github.com/sle118/squeezelite-esp32/files/6822692/airplay_synology.log)
[monitor output of ~15 seconds played through iPhone](https://github.com/sle118/squeezelite-esp32/files/6822695/airplay_iphone.log)

Sorry I did not discover the monitor sooner.

philippe44, 2021-07-15:
So it's a sync problem and that's why you see so many dropped frames. AirPlay allows you to synchronize multiple players and I support that, but it seems that the Syno controller is sending crappy sync control information and forces a lot of sync alignment (=dropped or skipped frames)

fieldOfView, 2021-07-15:
Sync seems to be an issue, but not the only issue. I modified https://github.com/sle118/squeezelite-esp32/blob/7951c59d114b99465397b5c509801c34858b1933/components/squeezelite/decode_external.c#L192 to always break, which in my understanding ignores all attempts to sync. This prevents the messages about skipped frames in the monitor, but the crackle is still there.

I am sort of ok with calling this a bad implementation of Airplay on the Synology side of things.

philippe44, 2021-07-15:
I was about to change something around that but if the crackles are still here, then it really looks like a codec problem on the syno side then

fieldOfView, 2021-07-15:
I think the sync issues might be a symptom, not the cause. I'll try to find out what sort of airplay stack syno uses, and try to reproduce it.

philippe44, 2021-08-29:
I'll close that for now

closed 2021-08-29 by philippe44