sle118/squeezelite-esp32 #92: Build with IDF v4.2

definable, 2021-04-25:
I am trying to build using IDF v4.2 (Eclipse/Windows) 
Ran into the following error: Unknown CMake command "esptool_py_flash_project_args"

When looking into squeezelite.cmake I see some lines commented out staring with: ## IDF-V4.2+

- Where does 'esptool_py_flash_project_args' refer to ? 
- It it possible to build with IDF v4.2
- If so, do I need to comment some lines and uncomment out the ones starting with ## IDF-V4.2+

Any suggestion or help woul be appriciated.

shaffenmeister, 2021-04-26:
Pleased refer to the manual:
https://github.com/sle118/squeezelite-esp32#manual-install-of-esp-idf

sangtraceur, 2021-05-08:
I have the same problem:
```
-- Adding new build target (from build folder): ninja squeezelite-flash
CMake Error at squeezelite.cmake:7 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".
Call Stack (most recent call first):
  squeezelite.cmake:124 (___register_flash)
  CMakeLists.txt:11 (include)
```
I've installed idf like you wrote in readme file

shaffenmeister, 2021-05-11:
> I've installed idf like you wrote in readme file
Did you source export.sh accordingly?
Did you run idf.py menuconfig or copy build-scripts to sdkconfig?
Did you use a different version of esp idf from the same terminal? ("spoiled" environment...)

MichaelPNolan, 2021-06-13:
I've been at this for a couple of days - i get further and further but ... can't get it to work. I'm using IDF4 linked in the guide. I'm copying the sdkconfig from the builds over. I'm using make not idf.py ... which doesn't work - it thorows this esptool_Py_flash_project_args command bs error.
Make gets almost all the way there and then is_recovery_running undeclare malloc function error

sle118, 2021-06-29:
@MichaelPNolan real life has caught on pretty bad on me lately, including the real work I do for a living, so I have had a bit less time for your past time project these past few weeks.

The primary mission was to build a system that would be very stable in a variety of configurations, and that goal was achieved.

Hopefully I'll catch up later this summer, or someone will send a pull request with updated instructions

closed 2021-06-29 by sle118

MichaelPNolan, 2021-06-29:
I hear you GL and thx - np