sle118/squeezelite-esp32 #94: GPIO control through squeezeserver CLI

Phil1pp, 2021-04-26:
First I want to say thank you for this awesome project.
Impressive how much functionality you got out of a cheap ESP32.

I integrated the ESP directly into an old AVR without WIFI capability.
Soldered the ESPs SPDIF output to the AVRs input.
Using one of the GPIOs to control the AVRs power.
Now I want to use additional GPIOs to control some other functions of the AVR remotely through squeezebox CLI.

My question: How can I send my own commands through the slimprotocol from squeezeserver CLI to the ESP32?
What's the easiest way?

sle118, 2021-05-10:
What do you have in mind in terms of use cases? I'm not sure if slimproto can be used to control gpio.

Phil1pp, 2021-05-11:
Well I would like to control the AVRs input selection (Tuner, TV, Squeeze) remotely and report some states of the AVR back to my home automation system.
I already tried it through MQTT, but this takes a lot of resources of the ESP and causes issues once Bluetooth is activated.
Therefore, I would like to use the slimprotocol and extend it by custom commands.
I just need a way to send commands to the ESP using the protocols that are already implemented.

Do you think this is somehow possible?

philippe44, 2021-05-19:
I assume you'd want to read GPIO as well? It's not super-complicated to do, I can do something with no verification at all, so it will be your responsibility to not screw up with GPIO you read/write

Phil1pp, 2021-05-20:
That would be great.
I don't want you to invest too much of your time on it.
Basically need a communication channel to send commands (and if possible show status) from the LMS webinterface to the ESP.
The GPIO digital and analog read/write on the ESP I can handle.

Phil1pp, 2021-08-30:
Do you already have an idea how this could be implemented?

philippe44, 2021-08-30:
it's just a new message to be sent to the player, these 4-bytes code, like what its being done for the Boom for i2c
```
# CLI I2C Example:  Set volume
# main volume control is at i2c address 47 (2f)
# Set volume to -20 dB.  0dB == 0x080000
# -20 dB == 0.1 linear.  
# 0x00800000 * 0.1 = 0x000CCCCC
#
# To send this volume command to a player called 'boom' do the following from the CLI:
#     boom boomdac %2f%00%0c%cc%cc    <- -20 dB
#     boom boomdac %2f%00%0c%cc%cc    <- back to 0 dB
sub boomI2C {
	my $request = shift;
	
	my $log = logger('player.firmware');
	
	# get the parameters
	my $client     = $request->client();
	my $i2cbytes   = $request->getParam('_command');

	if (!defined $i2cbytes ) {

		$request->setStatusBadParams();
		return;
	}

	my $data  = pack('C', 0x04);

	$data .= pack('C', length($i2cbytes));
	$data .= $i2cbytes;
	
	my @d = unpack("C*", $data);
	my $msg = "Sending data to i2c bus :[";
	
	foreach my $d (@d) {
		$msg .= sprintf("0x%02x ", ($d & 0xFF));
	}
	
	$msg .= "]";
	
	main::DEBUGLOG && $log->debug($msg);
	
	$client->sendFrame('bdac', \$data);
	
	$request->setStatusDone();
}
```

sle118, 2022-05-23:
Closing since there has not been any activity for a while