sle118/squeezelite-esp32 #97: Compile error with IDFv4.0

rsm123, 2021-04-28:
I'm getting compile error on manual install.

/Users/rsm/esp/esp-idf-v4.0/squeezelite-esp32/components/platform_config/platform_config.c: In function 'malloc_fn':
/Users/rsm/esp/esp-idf-v4.0/squeezelite-esp32/components/platform_config/platform_config.c:75:15: error: 'is_recovery_running' undeclared (first use in this function)
  void * ptr = is_recovery_running?malloc(sz):heap_caps_malloc(sz, MALLOC_CAP_SPIRAM);
               ^~~~~~~~~~~~~~~~~~~
/Users/rsm/esp/esp-idf-v4.0/squeezelite-esp32/components/platform_config/platform_config.c:75:15: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [platform_config.o] Error 1
make: *** [component-platform_config-build] Error 2

I'd appreciate any help to resolve this.

rsm123, 2021-04-28:
issue got resolved

closed 2021-04-28 by rsm123

MichaelPNolan, 2021-06-13:
i have this problem ... would like to know

rsm123, 2021-06-15:
> i have this problem ... would like to know

Don't remember, i think I selected previous IDF version.

MichaelPNolan, 2021-06-15:
Thanks - yes, that's often a big part of it.