sle118/squeezelite-esp32 #100: Bluetooth Class of Service

wizmo2, 2021-05-09:
I've been testing an AudioEngine HD3 speaker.  It's a 5.0 aptX HD but reports as an Audio device rather than the Render class.
I modded the esp_app_main,c
`761    else if (!(esp_bt_gap_get_cod_srvc(cod) & (ESP_BT_COD_SRVC_RENDERING | ESP_BT_COD_SRVC_AUDIO)))`

I get some weird errors when booting,

> I (15259) bt_app_source: Found BT Sink Device: Audioengine HD3 rssi is -79
> I (15259) bt_app_source: Found our target device. address 00:22:d9:00:1a:66, name Audioengine HD3
> I (15279) bt_app_source: Discovery completed.  Ready to start connecting to Audioengine HD3. 
> I (16739) bt_app_source: a2dp connecting to Audioengine HD3, BT peer: 00:22:d9:00:1a:66
> E (16739) BT_APPL: reset flags
> E (19719) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
> W (20069) BT_APPL: new conn_srvc id:18, app_id:0
> [00:38:49.877661] stream_sock:582 header: GET /stream.mp3?player=84:cc:a8:0d:3b:b0 HTTP/1.0
> ..... 
> [00:38:50.679362] write_cb:141 setting track_start
> [00:38:50.680742] resample_newstream:125 disable resampling - rates match
> [00:38:50.681693] process_newstream:124 processing: inactive
> E (23939) BT_BTC: handle_rc_connect Connect failed with error code: 2
> I (26739) bt_app_source: Output state is OUTPUT_BUFFER, Checking if A2DP is ready.
> I (26739) bt_app_source: a2dp media ready, starting playback!
> W (26749) BT_APPL: new conn_srvc id:18, app_id:1
> I (26779) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b
> 
> I (26789) bt_app_source: a2dp media started successfully.
> [00:38:55.448116] _output_frames:65 start buffer frames: 462848
> [00:38:55.459652] _output_frames:150 track start sample rate: 44100 replay_gain: 0
> [00:38:55.473667] equalizer_open:51 equalizer initialized 0
> W (26809) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
> W (26809) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 0

but then it works fine .

sle118, 2022-07-28:
Note to self:

Line that needs to be changed: https://github.com/sle118/squeezelite-esp32/blob/36f18fc069f01df3c600a28a68112f5f4407e34a/components/driver_bt/bt_app_source.c#L675

wizmo2, 2022-11-25:
I've submitted a PR #204 for this, along with a mod to help improve support for more bt speakers

sle118, 2022-11-25:
Closing this. I'll release a build today

closed 2022-11-25 by sle118