sle118/squeezelite-esp32 #105: Telnet input can cause restart

TheEclectic, 2021-06-09:
Trying to use telnet connection to monitor squeezelite-esp32 on a WROVER module (debugging connection/state problems trying to use a Bluetooth speaker). Noticed that sending an escape character caused squeezelite to restart. Problem code is in telnet.c function process_received_data() 'scrub from any escape command' processing.
I would hazard a guess that 'n' is supposed to be '\n' and it is  looking for the end-of-line; however, the telnet connection is typically reading a character or two at a time (from handle_telnet_conn()) rather than line-by-line so oftentimes all there is in the buffer is an escape character, 'size' is 1, and there isn't any newline in the buffer. The code doesn't respect 'size' and increments up through memory looking for an 'n' and will cause size to be decremented through zero causing size to become the unsigned representation of some negative number. The subsequent memcpy overwrites goodness knows what.

The behaviour of reading and processing a couple of characters at a time also means that you can't enter commands unless you are an Olympics level typist.

sle118, 2021-07-01:
I'm going to try having a look later in July when time permits. Thank you for reporting, and for the "Olympics level typist" as it made me smile :)

philippe44, 2021-07-04:
See https://github.com/sle118/squeezelite-esp32/commit/9cb18fa9806a8f0e25b29e1896935cb7e681d3c9

sle118, 2021-07-10:
I think I'm going to close this for now since Philippe pushed a fix. Feel free to reopen as needed

TheEclectic, 2021-07-10:
The change doesn't fix the code, other than detecting the newline should there happen to be one.

I would be happy to submit a pull request to fix process_received_data() myself if anyone can tell me what the function is actually trying to do.

* Why is '\e' a special case?
* Why isn't it a problem if there is a '\e' in the buffer other than at buffer[0]?
* Can there be multiple lines in the buffer? If so, should the code execute them in turn or silently drop everything after the first.

philippe44, 2021-07-10:
I don't know what the function wants to do. All I did was eating the chars following an \e up to end of buffer or \n, which I think was the original intend.

TheEclectic, 2021-07-24:
I didn't realise that further changes had been made to the code since https://github.com/sle118/squeezelite-esp32/commit/9cb18fa9806a8f0e25b29e1896935cb7e681d3c9 to protect from wandering off the end of the buffer until I tried to merge with my own re-write.

I am happy for this issue to be closed as it no longer causes a restart.

closed 2021-08-14 by philippe44