sle118/squeezelite-esp32 #108: Reboot Mute issue

danielschramm, 2021-06-11:
Hello,

I have a system using a PCM5102A I2S DAC.
It is only connected via 3 PIN I2C and Mute.
No other hardware.
Firmware: fw: I2S-4MFlash.16.689.master-cmake

The Playback is working but when the Server is in play mode (Online Radio station) and the device reboots the Audio output does not recover.
Pressing Pause and Play again on the server does not change the behavior.
The Server shows play mode but no audio output.
Pressing power off on the server and play again reactivates the output.

The behavior seems independent of using the Mute connection or just pulling mute to VCC on the PCM5102A. 
I'm not exactly sure what's causing this issue. 

Another System is using AC101 and I2S-4MFlash.570.master-cmake. It shows the same behavior.
So I do not think that it is DAC related but has something to do with Client Server Handshake during startup.
Logitech Media Server Version: 7.9.1

(As I remember, a reboot with Squeezelite clinet on linux does recover after restart.)

Thanks,

Daniel

philippe44, 2021-06-30:
First time I've seen that. Does you DAC requires initialization?

danielschramm, 2021-06-30:
> First time I've seen that. Does you DAC requires initialization?

That was also my first Idea but the AC101 has the same issue. That's why I think it might be something regarding the handshake with the server.
I will later recheck with a linux and esp32 client and can do some wireshark captures if they really behave different.

philippe44, 2021-06-30:
But the server does not do much, I mean it's either a wrong stream message / aude command. You'd need to take log of slimproto and squeezelite-esp32. Having said that, this is a corder case as hopefully, device does not reboot often...

danielschramm, 2021-06-30:
For me the ESP based clients boot really often since they are mobile clients and are powered off and on a lot that's why I discovered the issue. If the stream is not stopped properly on the server they are muted after power on but the server shows them as if they are playing.
The linux based clients are here are usually always on integrated in OpenWRT Access-Points.

I will create as much information on how to reproduce this as possible.

danielschramm, 2021-07-04:
[linux_client.zip](https://github.com/sle118/squeezelite-esp32/files/6759669/linux_client.zip)
[esp32_client.zip](https://github.com/sle118/squeezelite-esp32/files/6759671/esp32_client.zip)
Hello, 
I just compared Linux Squeezelite v1.9.1-1130 and ESP32 fw: I2S-4MFlash.16.689.master-cmake, mode: Squeezelite-ESP32.

RadioStation: 
Titel:	FFH Die 90er (Deutschland)
URL:	http://opml.radiotime.com/Tune.ashx?id=s97089&formats=aac,ogg,mp3,wmpro,wma,wmvoice&partnerId=16&serial=977e69e221e1df668964da983ff434f5

Uploaded you can find the Wireshark Captures. Both Clients were hard reset.
https://github.com/danielschramm/squeeze_logs

Linux resumes playback, ESP32 not.

I will also generate Logs soon.

Thanks,

Daniel

philippe44, 2021-07-29:
Oh, it seems you have not installed the LMS plugin that goes with the firmware. As a result, upon reboot, it connects to LMS with an ID that is rejected (DSCO) and then it retried with a mroe "conventional" ID, but meanwhile I'd guess that resume is cancelled by LMS.

danielschramm, 2021-07-29:
Hello Philippe,

I did not install the plugin since I had no Idea that it is needed for my usecase without a display:

The Documentation just says:
"To use the display on LMS, add repository ....."

I will retry with installed addon later today.

Thanks, Daniel

closed 2021-08-14 by philippe44