sle118/squeezelite-esp32 #109: 10 band Equalizer

rsm123, 2021-06-13:
Thanks for this wonderful software! I'm trying to understand how to use the 10 band Equalizer, especially while streaming  Airplay or Bluetooth. Apologies if I miss anything but I could not find this info in Readme or the web interface.

sle118, 2021-06-29:
Currently, the equalizer only works when connected to the LMS. To get it to work when in "offline mode" persistence needs to be implemented.

philippe44, 2021-06-30:
@sle118: please review the use of nvs_xxx. I can't use platform_config as I need blob but I don't remember the differences with nvs_xxx. Is it the delayed write?

sle118, 2021-07-01:
> @sle118: please review the use of nvs_xxx. I can't use platform_config as I need blob but I don't remember the differences with nvs_xxx. Is it the delayed write?

Funny enough I had edited your comment instead of replying (didn't know this was possible). So here was my response

I have reviewed the code, and yes the difference is the timer decoupling that allowed saving to Flash from a psram stack without smashing the cache. Let's keep the code as is for now, unless you think we're going to cause cache issues. I will update the method later and implement a mechanism that will allow using the config subsystem by covering to and from a hex string internally, or with a better approach

philippe44, 2021-07-01:
There is no risk at all for that to be used in PSRAM. You don't even need to worry about changing that then

sle118, 2021-07-02:
@rsm123 once the equalizer has been configured in LMS, it should now be saved and work with Bluetooth thanks to @philippe44 with commit https://github.com/sle118/squeezelite-esp32/commit/48e8525ba97657c155167095a7f0361146c64ebb

I'll close this for now

closed 2021-07-02 by sle118

larryqiann, 2025-05-13:
Hi,

Unsure where to put this. I'm using the latest build of Squeezelite-ESP32 and the newest Lyrion Music Server with the SqueezeESP32 plugin enabled in settings, but I can't find the equalizer option. What's the procedure for finding and using this functionality? 

Thanks!

larryqiann, 2025-05-13:
[image]

I've installed this and restarted LMS already. I'm assuming I should be looking under Player > Audio settings but I don't see the options.

Thanks for your quick reply!

philippe44, 2025-05-13:
There are some rare case that I've not been able to identify where the esp32 device is not proprely detected. Try to erase the player using Michael's plugin (can't remember the name, but it's made to erase inactive devices) and connect it again.

larryqiann, 2025-05-13:
I've just used the ESP web installer tool to reflash and erase the device, as well as factory resetting LMS. 
I do have a menu called "ESP32 settings" in the menu, but there is no equalizer config options under it. Could you confirm what the control should look like on a working setup? (Have tried both default and material skins)

[image]

philippe44, 2025-05-13:
oh, are you using the 32 bits version ? There is no equalizer then

larryqiann, 2025-05-13:
I'm using the I2S-4MFLASH-16 variant, so that should be the 16 bit version if I'm remembering correctly. I'm trying to reset/connect it to a Windows LMS instead to see if the problem persists. (is there an easy way to unpair the player without having to erase it and reflash? I'm having a hard time getting it to connect even after erasing)

philippe44, 2025-05-13:
Reflashing does not do anything for that issue. The only thing that might help is to remove the player form LMS

larryqiann, 2025-05-13:
What is the easiest way to remove a player from LMS? I couldn't get the player to connect to Windows at all, but the same problem is experienced on another Mac.

philippe44, 2025-05-14:
There is on plugin, made by Michael Herger and whose name escaped me now that does that. Look in the plugin list, it will be obvious when you see the name.