sle118/squeezelite-esp32 #111: No Sound on ESP32-A1S

Diddlik, 2021-06-29:
Hello,
I use the board "Squeezelite ESP32-A1S". It worked great with flashing, LMS server is running. Unfortunately, I get no sound, neither from the headphone output, nor speaker connection. I have not done any extra configuration.

Diddlik, 2021-06-29:
I connected via serialmonitor, than send a new track to esp32, it seems communication is working fine, but without sound.
[image]

sle118, 2021-06-30:
These esp-a1s boards have dip switches. I would start with making sure they are configured properly. Other than that, it is critical to Flash the a1s binary, since it is preconfigured for that hardware. You could also visit the forum and see if others have had this before

Diddlik, 2021-07-01:
I flashed with a1s binary. But could not find any information for dip switches configuration :(

philippe44, 2021-07-02:
Another user on forum (unless it's you...) is having the same issue with that board and it looks like the board is faulty. Have you tried any other SW to produce sound?

NB: I have one of these boards, I just checked the latest build and it works

Diddlik, 2021-07-03:
I did the same test and I god the same results...Maybe my board is faulty too. I will test asap wit hnew one.
Thank you

closed 2021-07-03 by Diddlik