sle118/squeezelite-esp32 #112: Setup for  ESP32-WROVER + I2S DAC

Diddlik, 2021-07-25:
I have a question of understanding how to best implement it.

I have a RAW module ESP32-WROVER + I2S DAC.
As DAC I have PCM5102 I2S DAC. Everything works so far. The sound comes both from headphones and from pins (L/R). The headphone output is ok, in the headphones really good sound and loud enough. When I connect the speakers to the pins it is extremely quiet. Do I understand it correctly that it should still be connected an amplifier? If so, unfortunately I have not found in the doc to do so.

shaffenmeister, 2021-07-26:
Yes, you should connect an amplifier as your board provides a line-out output only via both the headphone jack and the L/R pins.

Diddlik, 2021-07-26:
Ok. thanks!

closed 2021-07-26 by Diddlik