sle118/squeezelite-esp32 #113: Reconnect to well known WiFi access point if the connection is lost

ProASV, 2021-08-17:
It would be good if the esp32-wifi-manager do periodic attempts to restore the lost connection (after router reboot for example).

philippe44, 2021-08-29:
It does (I just tried). When connection is lost and depending if a server has been set explicitly, there is a certain amount of retries to either the previous server and then a search. After a while it reboots and may then enter AP mode if WiFi network is still not found. So what you might observe is that if your network is down for a while and after the reboot, then it is in AP mode.

ProASV, 2021-10-04:
Boot time of my router (asus RT-AC68U) is something like 1.5 - 2 minutes (yes, that's long enough). This is time from switching power on to raise the access point. So, every time then I switch off or reboot router I have to switch off/on squeezelite-esp32 then the access point is available. Otherwise I see an access point squeezelite-esp32  on the air and it stable without reconnect attempts to my network.

philippe44, 2021-10-04:
This is a fair request and I can do a hack to make it work, but @sle118 agreed to try to have a look and this is his realm in our software, so he would do something that would be better aligned with espressif's way (it can do softAP *and* STA at the same time)

philippe44, 2021-10-05:
I've pushed something here https://github.com/sle118/squeezelite-esp32/commit/13294ddd0c9fa8db7db0368b7d96b5265c778406. @sle118, let me know if this works for you. Seems to be fine in the tests I've done, but I've not created a release

philippe44, 2021-10-07:
See build #725

closed 2021-10-07 by ProASV