sle118/squeezelite-esp32 #115: Update from I2S-4MFlash from 32.710 -> 32.719 gives Bad-Request

sorryusernameisalreadytaken, 2021-09-02:
I try to update my ESP32-Wrover to the lastest firmware.

So success. Not with online update nor local bin flash. 

See here:
https://i.imgur.com/xrhX4z3.gif

robho, 2021-09-05:
I think the recovery image that you're running is too old. It doesn't support local file uploads and doesn't accept https (because github has a new certificate I think). Your options are:
1. Install from an http URL
2. Re-flash recovery with a more recent version

sorryusernameisalreadytaken, 2021-09-07:
> I think the recovery image that you're running is too old. It doesn't support local file uploads and doesn't accept https (because github has a new certificate I think). Your options are:
> 
> 1. Install from an http URL
> 2. Re-flash recovery with a more recent version

Awesome! Just a simple https to http change in the github does not work, but then I download ESP-Download-Tool v3.8.8 and flashed the latest recovery from this [file](https://github.com/sle118/squeezelite-esp32/releases/download/I2S-4MFlash.32.719.master-cmake/squeezelite-esp32-master-cmake-I2S-4MFlash-32-1.719.zip).

Perhaps it is cool for someone to see how:
[image]

closed 2021-09-07 by sorryusernameisalreadytaken