sle118/squeezelite-esp32 #116: opus_decompress error when playing from Youtube

Phil1pp, 2021-10-05:
When I try playing song from youtube through the youtube plugin it often stops playing after a few seconds.

The log says something about opus decompress error:

> [22:14:08.202352] output_thread_i2s:449 Output state is 1
> [22:14:08.253689] opus_decompress:153 setting track_start
> [22:14:08.543785] _output_frames:65 start buffer frames: 24648
> [22:14:08.546413] _output_frames:150 track start sample rate: 48000 replay_gain: 0
> [22:14:08.546863] output_thread_i2s:449 Output state is 2
> [22:14:15.834219] opus_decompress:244 op_read error: -136
> [22:14:15.834949] decode_thread:100 decode complete
> [22:14:22.029132] slimproto_run:808 output underrun
> [22:14:22.045018] output_thread_i2s:449 Output state is 0
> [22:14:32.674232] decode_flush:251 decode flush
> [22:14:32.674803] output_flush:447 flush output buffer
> [22:14:33.494125] codec_open:279 codec open: 'u'

I am not sure if this is caused by the Esp or the plugin and how to resolve.

philippe44, 2021-10-21:
Yes, the esp32 can only address 4MB os PSRAM, so you're pretty  much eating everything with buffers there :-)

Now, if you are not experiencing crashes with an increased buffer size, I might have alignment issues that happen very rarely and even much less when buffer is larger. I'll re-read code a bit

sle118, 2021-12-15:
Closing after inactivity

closed 2021-12-15 by sle118