sle118/squeezelite-esp32 #120: Buffer management

shark8594, 2021-10-19:
Hi,

First off great project!, I tried it on a custom audio board with 8MB PSRAM and ES8388 DAC and it works flawlessly. For my application, I expect that there would be packet loss from time to time due to fluctuation in the wifi RSSI. I think by default the rtp buffer seems to hold 2s worth of audio and also the max latency seems to be tuned to 2s. I would like to increase this to 8-10s and I tried increasing the #define BUFFER_FRAMES in rtp.c by a factor of 4 but this did not increase the buffered audio to 8s, it was more like 3.5s. Beyond a factor of 4 the ESP seemed to crash. I had a few questions regarding this:
1. Is it possible to buffer 8s-10s worth of audio on the ESP (latency is not a concern)
2. In case there is a network outage for 3s and the ESP sends a resend request, would the source have these frames or would they already be discarded?. I see in rtp.c that we do not even send resend requests if the seq_no(last,first) range is too high.
3. What knobs should I tune to increase the amount of audio data buffered (increasing OUTPUTBUF_SIZE did not help, I tuned the wifi  buffers and LwIP stack for maximum throughput)

Thanks a lot!
Log below:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7072
ho 0 tail 12 room 4
load:0x40078000,len:13772
load:0x40080400,len:4532
entry 0x400806a0
I (70) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.3-dirty 2nd stage bootloader
I (42) boot: compile time 22:10:11
I (42) boot: Enabling RNG early entropy source...
I (47) qio_mode: Enabling default flash chip QIO
I (53) boot: SPI Speed      : 80MHz
I (57) boot: SPI Mode       : QIO
I (61) boot: SPI Flash Size : 4MB
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (83) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (91) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  3 recovery         factory app      00 00 00010000 00140000
I (106) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (113) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (121) boot: End of partition table
I (125) boot: Defaulting to factory image
I (130) boot_comm: chip revision: 1, min. application chip revision: 0
I (137) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x5c488 (377992) map
I (267) esp_image: segment 1: paddr=0x0006c4b0 vaddr=0x3ffbdb60 size=0x03b60 ( 15200) load
I (273) esp_image: segment 2: paddr=0x00070018 vaddr=0x400d0018 size=0xbbd08 (769288) map
0x400d0018: _stext at ??:?

I (521) esp_image: segment 3: paddr=0x0012bd28 vaddr=0x3ffc16c0 size=0x007ec (  2028) load
I (522) esp_image: segment 4: paddr=0x0012c51c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/freertos/xtensa_vectors.S:1778

I (528) esp_image: segment 5: paddr=0x0012c924 vaddr=0x40080400 size=0x161d8 ( 90584) load
I (571) esp_image: segment 6: paddr=0x00142b04 vaddr=0x400c0000 size=0x00068 (   104) load
I (585) boot: Loaded app from partition at offset 0x10000
I (585) boot: Disabling RNG early entropy source...
I (585) psram: This chip is ESP32-PICO
I (589) spiram: Found 64MBit SPI RAM device
I (594) spiram: SPI RAM mode: flash 80m sram 80m
I (599) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (607) cpu_start: Pro cpu up.
I (610) cpu_start: Application information:
I (615) cpu_start: Project name:     SqueezeAMP
I (620) cpu_start: App version:      local.500.cmake-master
I (627) cpu_start: Compile time:     Oct 17 2021 16:11:22
I (633) cpu_start: ELF file SHA256:  f9ac4d24ad7c2b9a...
I (639) cpu_start: ESP-IDF:          v4.0.3-dirty
I (644) cpu_start: Starting app cpu, entry point is 0x40081924
0x40081924: call_start_cpu1 at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/esp32/cpu_start.c:272

I (0) cpu_start: App cpu up.
I (1151) spiram: SPI SRAM memory test OK
I (1152) heap_init: Initializing. RAM available for dynamic allocation:
I (1152) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[... 373 lines cut ...]
[00:02:25.629200] output_thread_i2s:449 Output state is 0
[00:02:25.631590] output_thread_i2s:512 Restarting I2S.
[00:02:25.634215] equalizer_open:71 equalizer initialized 0
C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/freertos/tasks.c:683 (xTaskCreateStaticPinnedToCore)- assert failed!
abort() was called at PC 0x40099d06 on core 1

ELF file SHA256: e966a361cb757a9f

Backtrace: 0x40095ea3:0x3fac34d0 0x40096211:0x3fac34f0 0x40099d06:0x3fac3510 0x40104d8a:0x3fac3550 0x4010312b:0x3fac35f0 0x40103791:0x3fac3780
0x40095ea3: spi_flash_chip_generic_erase_chip at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/spi_flash/spi_flash_chip_generic.c:89
 (inlined by) spi_flash_chip_generic_erase_chip at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/spi_flash/spi_flash_chip_generic.c:80

0x40096211: spi_flash_chip_generic_set_io_mode at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/spi_flash/spi_flash_chip_generic.c:352

0x40104d8a: esp_phy_rf_deinit at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/esp_wifi/src/phy_init.c:208 (discriminator 3)

0x4010312b: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/nvs_flash/src/nvs_storage.cpp:357
 (inlined by) nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/nvs_flash/src/nvs_storage.cpp:266

0x40103791: nvs::HashList::find(unsigned int, nvs::Item const&) at C:/Workspace/sqeezelite_sandbox/esp-idf-v4.0.3/components/nvs_flash/src/nvs_item_hash_list.cpp:107

I (2578) esp_core_dump_uart: Press Enter to print core dump to UART...
I (2585) esp_core_dump_uart: Print core dump to uart...
I (2591) esp_core_dump_common: Found tasks: (23)!
================= CORE DUMP START =================

philippe44, 2021-10-21:
Just to make sure I understand, you're talking about the AirPlay part of that project, right? The main feature is around LMS/Squeezelite and AirPlay is something I added after, not as a hack, but as a quick extension. On AirPlay1, 2 is the very usual buffer time and I'm a bit surprised you're experiencing a lot of frame drops.

philippe44, 2021-10-21:
Understood - This is a more complicated request as in AirPlay, the buffering is semi-negotiated between the client and the server but quite often the server (your iPhone imposes it). I'll see what I can do

sle118, 2023-04-01:
is this still needed, or shall we close the issue? 
@philippe44 ?

philippe44, 2023-04-01:
I looked at it and I forgot that for memory optimization the AirPlay buffer is stolen from the common output buffer which is artificially shrunk during AirPlay playback. The output buffer is what is the yy set on the command line with -b xx:yy. 

The numbers are as follows:

- Output buffer reduced from yy to 120% of 2 seconds = 420kB
- Memory available for AirPlay is yy-420kB. By default it's is 150% of 2s of buffers chunk of 352 samples each, so 375 chunks. Each chunk is 24+352*4=1432 bytes, so total is 537k. 
- So today we already need an yy of above 1MB. 

If you increase the number of buffer chunk as the OP did, you will exceed the yy and that will not end well.

Best we could do is likely another 1MB so it is about another 4s. Today it's a statically set number, I'll see if I can make it dynamic

philippe44, 2023-04-01:
Done, can be closed

closed 2023-04-02 by philippe44